djuseeq/Ch376msc #1: Can't get SPI to work

BleuLlama, 2019-05-08:
Hi. I have my CH376 module connected to an Arduino Leonardo (ATmega 32u4), with BUSY configured on D9, and CS on D10 (which is the defaults in the example code).

When I run the SPI example code, I can see that it is communicating as it enables USB detection; when i have my thumb drive plugged in, the LED on my CH376 breakout board is lit, when i remove it, it is not lit.

I have used this same drive and same interface board in UART mode, which worked fine, but my needs changed and I need to use SPI mode.  When I had it working in UART mode, I knew it was configured right when i sent the USB MODE command, which resulted in the LED being lit in this way...

So I know that the CH376 is receiving the data, but every example responds with "Attach flash drive first!"

BleuLlama, 2019-05-08:
Aha.  I think i figured it out. I had a microsd interface also on the SPI bus; both were connected to MISO, and your code mentions that the interrupt stuff only works if there are no other devices on the SPI bus.  I removed the microsd interface and now it is working.  I'll close the ticket. :D

closed 2019-05-08 by BleuLlama

djuseeq, 2019-05-09:
Hi, sorry for the late response. If you have one extra free pin on your Leonardo board then connect it to usb module's INT pin and when you create an instance of the class, use: 
`Ch376msc flashDrive(SS, BSY, INT); // chipSelect, busy, interrupt pin`
With that you can share the SPI bus. Thanks ;)