djuseeq/Ch376msc #3: CH376 Library MoveCursor issue

yaronze, 2019-09-17:
Hi,
i use the library with Arduino UNO reading a text file from disk on key (DOK).
while reading the file with example everything is great.
i am trying to read a section from the file using MoveCursor. i noticed that it works fine until a point where the byte address is close to 512 byte area in the file and so on - 1024, and multiplications of 512. at this point i am getting part of the correct data from the file and after 512 location it seems to get the first bytes of the file.
is it a bug?
am i doing something wrong?
thanks

djuseeq, 2019-09-17:
Hi. That is probably a bug.The sectorCounter variable is not updated during moveCursor call. I will look on it.
Thanks

yaronze, 2019-09-18:
Hi,
thank you!
it will be great if you will fix the bug and update the great library.

djuseeq, 2019-09-18:
It's corrected and updated. Thank you for your appreciation, it means a lot to me.
I'm closing this issue.

closed 2019-09-18 by djuseeq

djuseeq, 2019-09-18:
If you still have a problem with it, please reopen this issue.

yaronze, 2019-09-18:
Hi,
Thank you very much.
I made a little test and it seems work as it should.
Thank you for your fast response.
If i'll find any problem then i will report it again.
Have a nice day!