djuseeq/Ch376msc #5: Read .csv file from flash drive

yaronze, 2019-09-19:
Hi,
I am working with the updated CH376msc library.
I tried to read a csv (type of Excel) file from flash drive using the library example, but no success.
After i changed the file extension on flash drive to TXT - everything work just great.
A .csv file is actually kind of text file where data is comma delimited.
Why it is not possible to read other files that are not TXT type?
Is there any way that will enable read other file types?
Thanks

BleuLlama, 2019-09-19:
Make sure your filename is all capitalized.

djuseeq, 2019-09-19:
Hi,
as BleuLlama said, in a setFileName function write your file name and extension all capitalized

yaronze, 2019-09-20:
Hi
Thank you for your response.
In the SetFileName i wrote file name and extension all capitalized but still when file name is "MYFILE.TXT" it works great and when file name is "MYFILE.CSV" it is not working.
I double checked and both file exists on flash drive.

djuseeq, 2019-09-20:
I have tested and opened files with .CSV extensions without problem.I created 3 file on the flash drive, TEST1.CSV, test2.CSV and test3.csv . In all three cases i had to use capitalized filename and extension while calling setFileName function otherwise i got file not found error code (0x42). Try to insert a serial print while calling openFile function and let`s see what happens. Instead of
`flashDrive.openFile(); ` 
use 
`Serial.println(flashDrive.openFile(),HEX); `  and watch your serial monitor

yaronze, 2019-09-20:
Hi,
I checked again the file name and extension.
I added Serial.println(flashDrive.openFile(),HEX); and got 42 code in serial monitor.
While changing file extension to .TXT in code and in flashdrive - it works fine.
I do not know what am i doing wrong or if it a bug.

djuseeq, 2019-09-20:
42 code means, the ch376 cannot find a file specified. I think the problem is around your flash drive or file name. Don't use longer file name than 8 character + 3 character extension. e.g. JOHNDOE1.CSV . [https://en.wikipedia.org/wiki/Design_of_the_FAT_file_system#Directory_entry](url)
Flash drive must to be formatted in FAT32 or FAT16

yaronze, 2019-09-20:
I am using flash drive formatted FAT32.
The file name is 5.3, meaning 5 characters name and 3 characters extension as shold be.
i will try with another flash drive and continue testing.
Thank you for your responses and testings.

djuseeq, 2019-09-21:
Hi,
you can try to create a .csv file with the ch376 and then try to read it. In example code change everywhere the "TEST1.TXT" to e.g. "TEST1.CSV"

yaronze, 2019-09-21:
Hi
I created a .csv file.
It seems to have another problem - it mixing the text. file append seems to write the data not as it should.
i am writing string - 
     char adat[] = "Log,Time,Action,Step,Step Name,Status";
       flashDrive.writeFile(adat, strlen(adat));
When i opened the .csv file, some of the data is mixed up -
"Log,Time,Action,Step,Step Name,StatusLo"
Than i write some other data and it seems missing part of the chars.
I worked with same code with SD module and it works just as it should.
I made changes for working with CH376 library, and writing to file causing these problems. I cant understand why.

Log | Time | Action | Step | Step Name | StatusLo
-- | -- | -- | -- | -- | --

yaronze, 2019-09-21:
I also did what you suggested - i changed the file extension in the library example to .CSV.
The TEST1.CSV file was created ok. The text is in the file.
I can read the .csv file using command '3'.
It seems that the problem is to read a .csv file that was created by Excel. I'm not sure.

BleuLlama, 2019-09-21:
Is the file ASCII or UTF-8?  If it's UTF/unicode/16 bits/byte it might have
problems?

djuseeq, 2019-09-21:
> Hi
> I created a .csv file.
> It seems to have another problem - it mixing the text. file append seems to write the data not as it should.
> i am writing string -
> char adat[] = "Log,Time,Action,Step,Step [...]

Ahha, you forget the '\n' (new line) character. 
`char adat[] = "Log,Time,Action,Step,Step Name,Status\n";`
Try with that, and of course when you add a new record, put the new line char at the end

djuseeq, 2019-09-21:
> Is the file ASCII or UTF-8? If it's UTF/unicode/16 bits/byte it might have problems?

It can be or the MS excell add some extra data to his csv file

djuseeq, 2019-09-21:
I made a test with
`char adat[] = "Log,Time,Action,Step,Step Name,Status\n";`
`char adat2[] = "bla,bla,bla,bla,bla,bla\n";`
On screenshot1 you can see the terminal window
[image]
[image]
[image]

yaronze, 2019-09-21:
Yes.
Thats OK also for me :) Thank you.
The problem is at situation like this -
logString = String(logCount)+ ','+ strTime+ ',' + logAction + "," + String(logStep)+ "," + logStepName+ "," + stateErr + ",\n";
I convert the string into char array, all data is written in .csv file but no effect of "\n". so, data is written in serial and not line feed between the writings.

djuseeq, 2019-09-21:
Your code is public? It would be much easier to understand the problem if we could see the code. Probably something goes wrong when you convert the String object to char array, so I'm just guessing

yaronze, 2019-09-21:
Hi,
This is the code. It is part of big project sketch.
//********** LOG Process ******************
void proccess_log() {
  char adat[80];
  String logString;
  String stateErr = F("N");
    
  //call miliseconds to time conversion
  millis_to_time();

  if (fileError == true) {stateErr = F("Error");}
     
  logString = String(logCount)+ ','+ strTime+ ',' + logAction + "," + String(logStep)+ "," + 
 logStepName+ "," + stateErr + ",\n";

   if(flashDrive.checkDrive()) {
     //open log file for writing
     flashDrive.setFileName("ELOG.CSV");  //set the file name
     flashDrive.openFile();                   //open the file
     
     if (logCount ==1) {                       //Write log header in row 1.
       char adat[] = "Log,Time,Action,Step,Step Name,Status\n";
       flashDrive.writeFile(adat, strlen(adat));     //string, string length - write header into log file
     }
     flashDrive.moveCursor(CURSOREND);        //move the "virtual" cursor at end of the file
     logString.toCharArray(adat,logString.length());
     flashDrive.writeFile(adat, strlen(adat));     //string, string length - write data into log file
          
     flashDrive.closeFile();               //at the end, close the file
     logCount = logCount + 1;      //increment log counter  
  
 }else {                            // if the file isn't open, pop up an error
    Serial.println(F("error opening Elisalog.txt"));
  }
  
}

djuseeq, 2019-09-22:
Hi,
the problem is in your code. char adat[] array is declared twice, the String objects are good just because of their simplicity but they are memory eaters. I don't see how you handle the logCount variable, stored in the EEPROM? Convert [StringToArray](https://circuits4you.com/2018/03/08/arduino-convert-string-to-character-array/), read it carefully, particularly the "len" parameter.
`
void proccess_log() {
	char adat[80] = "Log,Time,Action,Step,Step Name,Status\n";
	int logStep = 1;
	static int logCount = 1;
	String logString;
	String stateErr = F("N");
	String strTime = "18:05:14";
	String logAction = "Start";
	String logStepName = "Test";
	//call miliseconds to time conversion
	logString = String(logCount)+ ','+ strTime+ ',' + logAction + ',' + String(logStep)+ ',' +
			logStepName+ ',' + stateErr + '\n';

	if(flashDrive.checkDrive()) {
		//open log file for writing
		flashDrive.setFileName("ELOG.CSV"); //set the file name
		flashDrive.openFile(); //open the file
		if (logCount ==1) {                       //Write log header in row 1.
			flashDrive.writeFile(adat, strlen(adat));     //string, string length - write header into log file
		}
		flashDrive.moveCursor(CURSOREND);        //move the "virtual" cursor at end of the file
		logString.toCharArray(adat,logString.length()+1);
		flashDrive.writeFile(adat, strlen(adat));     //string, string length - write data into log file
		flashDrive.closeFile();               //at the end, close the file
		logCount = logCount + 1;      //increment log counter
	}else { // if the file isn't open, pop up an error
		Serial.println(F("error opening Elisalog.txt"));
	}

}`
[image]

yaronze, 2019-09-22:
THANK YOU!!!
The problem writing .csv was in my code and you found it - the offset of '1' in length.
Thank you!
It works great.

Now i got to figure out why the .csv file gives 42h error code. I have a walk around to work with .TXT file, but prefer nice .csv.
Have a great day

djuseeq, 2019-09-22:
I'm glad I could help and it works. So then i close this issue.
Good luck.

closed 2019-09-22 by djuseeq