djuseeq/Ch376msc #7: Request

indramal, 2019-09-21:
First, I want to thank you for this library. I also used Scott C example code. Scott C code can create the file but can not read in computer. Then I found your library with improvement of Scott C codes. 

I saw your code added FAT file structure. Is it FAT32 file structure or other? I have not seen those types and size variable in this link https://en.wikipedia.org/wiki/Design_of_the_FAT_file_system. 

was it the problem in Scott C example code? (FAT file structure not included in Scott C example code). 

What are other possible extensions to save files in pen drive? (Other than txt and csv)

djuseeq, 2019-09-21:
Hi,
i'm glad you found it useful this library. Actually the ch376 chip handle the FAT file structure(FAT12 FAT16 and FAT32).This is hardcoded to the chip itself that means the NTFS support is not possible :( . I needed information about FAT file structure because i wanted to understand and handle the directory entry i get it from the chip. I used in these functions: rdUsbData(), writeFatData().So the ch376 do the hard job.I used just for plain text, because that what i needed for.

Fenix-125, 2019-09-29:
I have encountered similar problems to find out the right file structure.
And I want to share my solution.
This is the output of "**cfdisk**" and "**fdisk**" commands respectively (**Linux**):
[image]
[image]

Exialy the FAT formatting on Windows 10 did not match the necessary one, but the one (FAT32) on Windows XP does.

djuseeq, 2019-09-29:
Thanks for information, Fenix-125. That means who use Win10, they need to found some formatting tool to make a proper FAT partition on theirs flash drive. I'm on Linux,so i didn't faced with this kind of issue.

closed 2019-11-21 by djuseeq