djuseeq/Ch376msc #9: Possible to open other file types ?

gcharles81, 2019-09-30:
Hello thanks for this library 
Is it possible to open other file types ? or it will support only TXT files
thanks

djuseeq, 2019-09-30:
Hi, it's my pleasure . Actually yes, you can open , but probably it gonna be unusable. The CH chip will send a large pile of bytes and that somehow needs to be processed. If the goal is some data logging then best choice is CSV(Comma-separated values), it's easy to open in MS excel or LibreOffice. Which extension are you interested in? Here's an example to see what I'm talking about. 
[image]
[image]

luc-github, 2019-10-07:
hi sorry to jump in conversation but as topic is still open ;)
I did not dig in code yet but do you mean that reading binary like pictures may be a problem?
I would like to store text based files as well as binaries for images

djuseeq, 2019-10-08:
Hi, i haven't tried  to use for binary files. But that feature can be useful , i will experiment with it

djuseeq, 2020-10-29:
Hi @luc-github , maybe [this](https://github.com/djuseeq/Ch376msc/issues/35#issuecomment-717952129) can be useful for You.

closed 2020-11-03 by djuseeq

luc-github, 2020-11-03:
@djuseeq thank you - I am super late to integrate your lib - I will check that when on it