djuseeq/Ch376msc #10: [Documentation]Jumper position?

luc-github, 2019-10-07:
Nice lib thank you so much for your work, I got hard time to find clear doc about CH376s using SPI until I saw your lib. soo happy

You write jumper configure the communication way (SPI/Serial)
but picture always show same position
Is that normal?

FYI I plan to use it on ESP32 and report back and eventually do a PR to add ESP32 in library.properties if it is ok

djuseeq, 2019-10-08:
Hi, I'm glad you found it useful . If you mean H1 jumper position is same then yes, that's correct, the other position is for Parallel communication which one is not implemented in this lib. Okay, i'm waiting for your report.

luc-github, 2019-10-08:
ok thank you. I need to setup a prototype as  esp32 use 3.3v and it is not 5v tolerant on GPIO.

djuseeq, 2019-10-08:
Check the [datasheet](https://www.mpja.com/download/ch376ds1.pdf#%5B%7B%22num%22%3A61%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C0%2C382.596351%2C0%5D) for 3.3V setup. Probably you must to do some corrections on the module.

luc-github, 2019-10-08:
ok thank you

closed 2019-10-09 by djuseeq