djuseeq/Ch376msc #11: for Zero SAMD

gioreva, 2019-11-11:
SetGet.cpp:

#include "Ch376msc.h"
char *dtostrf (double val, signed char width, unsigned char prec, char *sout) {
  asm(".global _printf_float");

  char fmt[20];
  sprintf(fmt, "%%%d.%df", width, prec);
  sprintf(sout, fmt, val);
  return sout;
}

I removed a const  somewhere

gioreva, 2019-11-11:
#ifndef dtostrf
char *dtostrf (double val, signed char width, unsigned char prec, char *sout) {
  asm(".global _printf_float");

  char fmt[20];
  sprintf(fmt, "%%%d.%df", width, prec);
  sprintf(sout, fmt, val);
  return sout;
}
#endif

djuseeq, 2019-11-13:
Hi @gioreva , that means you have tested Zero SAMD and is it works with it? Thanks for testing.

gioreva, 2019-11-13:
Yes, tested and work.

Other things

CH376T   20pin, has not Busy Pin.
But enought 3uS delay.

```
void Ch376msc::spiReady(){
	// uint32_t msTimeout;
	delayMicroseconds(3);
	// msTimeout = millis();
	// while(!digitalRead(_spiBusy)){
		// if(millis()-msTimeout > TIMEOUT){
			// break;
		// }//end if
	// }//end while
}
```

vladkozlov69, 2019-11-13:
> CH376T 20pin, has not Busy Pin.

As far as I remember, there is a busy flag in the interface status data (read from command port) so it could be used for the T chip modification.

gioreva, 2019-11-13:
I read this in the datasheet:

③ SCM inquiry BZ pin waiting for SPI interface of CH376 is free, or directly delay TSC time (about
1.5uS); 

But if it is not enough, can you add at the library ?

djuseeq, 2019-12-02:
Hi, please check this [branch](https://github.com/djuseeq/Ch376msc/tree/SAMD) . It would be great if you can test it, also with the example files. If everything works fine then i will make a PR to the master branch. Thanks

gioreva, 2019-12-02:
I assemble new [board ](https://shop.gevaelettronica.it/en/arduino-compatible/63-gevino-tft-7.html)in a few weeks.

djuseeq, 2019-12-05:
I would totally skip using the BUSY pin. Stability problems have been encountered in communication without a BSY pin until now. If all goes well, I'll update the SAMD branch within days

gioreva, 2019-12-27:
Il tuo link non esiste più.
I have now assembled a prototype and I wanted to try.

djuseeq, 2019-12-27:
Hi and sorry, i`m still learn how to use Git, the MASTER branch is updated to 1.4.1, so you can try that one.

gioreva, 2019-12-27:
Can you give me an example of intensive testing ?
Create a large text file, or copy a large file to another.

djuseeq, 2019-12-27:
To create a large text file is easy to achieve, just change the `for(int a = 0; a < 20; a++)` to e.g. `for(int a = 0; a < 4000; a++)`(or larger number) in example file.How you want to copy a file?If You have 2 modules then easy read from module1 to buffer, write to module2 from buffer sequence. If you want to use just one module then ,Warning, pseudo code:
 open "a" file read to buffer save in to a variable the amount of readed bytes, close file "a", create file "b" write data from buffer, close file "b". Open file "a", use moveCursor(variable with previously readed data), read to buffer, close file "a" add to readed byte variable the new amount of data .....

djuseeq, 2019-12-27:
I think the second advice with one module is not so good, with that you will quickly wear off your flash drive, [see thiss section](https://www.mpja.com/download/ch376ds1.pdf#%5B%7B%22num%22%3A65%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C0%2C710.22748%2C0%5D) ,focus on CMD_FILE_CLOSE

gioreva, 2019-12-27:
Not Work
The previous version was not drive the CS, I had connected to the D45 SWDIO
The pin never change state.
Ch376msc flashDrive(45);

This version Crash SerialUSB driver.
Windows says "unknown USB devices"

I remember that I use SAMD CPU, Zero Arduino board.
It has the internal USB chip.

djuseeq, 2019-12-27:
Try with `Ch376msc flashDrive(CS, INT);` because some boards don`t like if i want to use MISO as Input pin or the SPI bus is shared with something.

gioreva, 2019-12-27:
Same, unknow device.

Ch376msc flashDrive(48,0);

gioreva, 2019-12-27:
Preview version work, I see and interact with the menu, but I always say, USB memory is not inserted and the pin 45 does not move.

djuseeq, 2019-12-27:
Previous version is still accessible [here](https://github.com/djuseeq/Ch376msc/releases/tag/1.4.0) . Please can you send me the exact module what you have? Please double check wiring

gioreva, 2019-12-27:
This version not compile.

https://pastebin.com/U6Yf45Qk

gioreva, 2019-12-27:
Wait, i forgot change at the head of this post.

gioreva, 2019-12-27:
Then, i have ZIP of workin version.
is   -->  v1.4.0 Sep 26, 2019 
With Change Top Here.
Work using some pin for CS, not for D45

djuseeq, 2019-12-27:
Try to use example file from the new version.  `void printInfo(const char* info)` is changed to `void printInfo(char info[])`

gioreva, 2019-12-27:
We do not need this test, I had already corrected that line on v1.4.0 before
The problem is that described.

djuseeq, 2019-12-28:
Ok, i think i get it on what You means. Is it `#ifndef dtostrf` ? Because that is defined automatically if You want to compile to SAMD architecture. See in Ch376msc.h file 
`#if defined(ARDUINO_ARCH_SAM) || defined(ARDUINO_ARCH_SAMD)
	#include "avr/dtostrf.h"
#endif`
I tried to compile to Zero and it works without any issue. Physically i tested on SAM(Arduino Due) board and it works flawlessly on SPI bus BUT just if i hook up one extra pin as INT. Can You compile it?
I don`t have Arduino Zero board, so im not familiar with it, but [here](https://www.arduino.cc/en/uploads/Main/ArduinoMKRZero-schematic.pdf) i can`t see it have pin 45 or 48 what You try to use as CS pin. How You mean  "This version Crash SerialUSB driver." ?
Thanks

djuseeq, 2019-12-29:
Is there any progress with the project?[This](https://cdn.hackaday.io/files/19363828426592/GEVA%20ARDUINO%20ZERO%20Schematic.pdf) is your board? So you working in a 3.3V environment, if you use the CH376 module from ebay then they are already have 3.3V voltage regulator so the IO pins are compatible with your system.Be aware on the module pin marked CS is not ChipSelect pin for SPI,this pin you have to pull to logic HIGH(3.3V) see the [picture](https://github.com/djuseeq/Ch376msc/blob/master/extras/JumperSelect.png). Regarding your schematic file the SPI bus is shared with SD card and with ICSP. Try to implement what ATMEL/Microchip [recommends](http://ww1.microchip.com/downloads/en/AppNotes/Atmel-2521-AVR-Hardware-Design-Considerations_ApplicationNote_AVR042.pdf#_OPENTOPIC_TOC_PROCESSING_d94e1039) and/or use pullup resistor for CH376 ChipSelect pin.

gioreva, 2019-12-29:
Tomorrow I try your advice.
I also want to see if it is the fault of the Digital 45 if it crash USB
My board is [this](https://shop.gevaelettronica.it/en/arduino-compatible/63-gevino-tft-7.html)  
Schema is similar to your link but not same.
I have soldered  CH376T  directly on the PCB ad there are 5V dedicatet regulator for power up memory stik. (Or memory stick work also at 3.3V ?)

I can send you schema if you want.

Normally, when you connect SAMD CPU at windows PC, it load "Arduino zero"  driver.
After loaded your code, windows do not load driver, USB enumerator says an unknown numer, unknown device.
Your code change some usb register on the CPU.

Adding this
#if defined(ARDUINO_ARCH_SAM) || defined(ARDUINO_ARCH_SAMD)
	#include "avr/dtostrf.h
	error_here
#endif
I receive error, then i think condiction are executed.

And renoving it i have not errors.
(Library and example, downloaded now)

gioreva, 2019-12-29:
[Schematic GEVINO TFT.pdf](https://github.com/djuseeq/Ch376msc/files/4008192/Schematic.GEVINO.TFT.pdf)
This schema.

djuseeq, 2019-12-30:
Hi, that is fine, 5V is needed for flash drive.Here is the list what i found, these would be good to review:
1. in your schema 1uF is used between RSTI pin and VCC, [here](https://www.mpja.com/download/ch376ds1.pdf#%5B%7B%22num%22%3A45%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C0%2C156.254846%2C0%5D) the datasheet recommend 100nF.
2. the RXD pin is connected to VCC. Leave it unconnected.
3. there is a 10nF capacitor between V3 pin and GND.The datasheet [here](https://www.mpja.com/download/ch376ds1.pdf#%5B%7B%22num%22%3A61%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C0%2C382.596351%2C0%5D) for 3.3V operation recommends to connect V3 pin to VCC
4. the chip require 12MHz +- 0.4% clock source on his XI XO pins,i'm not sure so please confirm, are you use ceramic resonator or crystal with built in capacitors or just the schematic symbol is ambiguous? Resonator i think is not enough accurate for this requirements, if crystal with capacitors then the extra external capacitors can make trouble for the oscillator
5. CS(SS) or chip select pin require an external pullup resistor because before the pin init done in the MCU(bootloader is starting or programming the MCU over ICSP?) the CH376 chip is in unstable state, it can receive some commands from MOSI pins what is not intended for him and make trouble.But i see you use the same pin as SWDIO, is it for debugging?Is it possible to use a dedicated pin for it? 
6. And finally the most important, INT# pin is not connected to the MCU. Please check the API [here](https://github.com/djuseeq/Ch376msc), if the SPI bus is shared and here is shared(CH376,AUX,RA-02,SD card,Ethernet,Shift register) then you must to use one extra MCU pin as INT input from the chip and use constructor `Ch376msc(spiSelect, interruptPin);` because otherwise the CH376 chip will pull down his MISO pin(MISO pin on the whole SPI bus) to LOW if it have some interrupt request.
Thanks

djuseeq, 2019-12-30:
Please confirm, on this board with the 1.4.0 library and with your modification everything works just fine?Or that was on prototype board with different pins and without any other SPI peripherals?
Thanks

gioreva, 2019-12-30:
Thank you for your advice

Truly a very long time to write the phrase 1,000,000 times, but in the end everything works perfectly.
Ch376msc flashDrive(45);

1, 2,3 Ok
4, it is Crystal Without Capacitor.  Similar to [link](https://www.mouser.it/datasheet/2/905/ndk_04232018_NX3225SA_STD-CSR-3-1324920.pdf)
5 I read that there is already a pullup in the chip.

6 Non ho più pin per connettere INT
C'è una soluzione alternativa ?
If you use Ch376msc flashDrive(45, -1) ?
When taking action on INT ? When I insert the key ?
It is an internal register that says to lower MISO when INT ?

djuseeq, 2019-12-30:
You are welcome and really sorry then i misunderstood you (it's my fault):( . Okay, than if you are sure in pullup (5.) then focus on(6.), if you don't have more free IO pin on the MCU then the things get more difficult, in SPI mode the chip can works in 2 mode, 1, to use MISO as interrupt request pin when CS is high(but this mode is not work's if the bus is shared), 2, with one extra IO pin as INT(when more devices hooked to the SPI bus like in your situation).The constructor expect a unsigned(positive) values.The chip send an interrupt request(INT pin goes to LOW) among others when we attach/detach a device and in case when done with a [command](https://www.mpja.com/download/ch376ds1.pdf#%5B%7B%22num%22%3A7%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C0%2C368.675029%2C0%5D) (see note Produce interrupt in column Output). I will check it is it possible to get it somehow on other way (polling,reading) some internal register but I can't promise anything.

gioreva, 2019-12-30:
Sorry for my english, I'm lost.
"write the phrase" I meant written on Flash, test write function.
But really, 1'000'000 loop has crashed CPU, PC, all.
1'000 has work well.

When i use SPI for read write on flash i do not use same for other.
SAMD is not multitasking and has not DMA
Now it works, I do not know how it will behave when I will use all of the devices, but I think it can work.
If INT is when he finished writing, I do not give problems.

djuseeq, 2019-12-30:
Ahha, okay, no problem,yes, it takes really long time, you can speed it up because the default com speed is 125kHz.I'm quite confused reading the datasheet regarding the maximum SPI clock, because mention 6MHz (maybe for the SD card??) and 24MHz. You can change the speed in Ch376msc.h file, edit the SPICLKRATE line. To use MISO as INT can be a problem if e.g. the MCU start communication on SPI bus with other device and during this process you attach/detach a flash drive because in this case the CH376 will pull down the whole MISO line(interrupt request) and make data corruption in the communication with the other device.

gioreva, 2019-12-30:
TFT now use 12Mhz SPI
The waveforms are very rounded, I do not think can work at 24Mhz on My PCB.
They are very long tracks, I have SPI 6 items on PCB.

You can add the SD commands ?
They are the same TFT
I think they're already in the Arduino IDE

SD.print
SD.println
ecc.

djuseeq, 2019-12-30:
Then you can try 6 or 8MHz. I need to experiment with the module, solder an sdcard holder to it. If i have a free time i will look into it.

gioreva, 2019-12-30:
Set 20 000 000 Mhz on .h
Oscilloscope says there are 12Mhz on clock

Byte read are wrong

************************
COMMAND7: List directory
************************

7@rqI : 27318 >>>	26.7 Kb

<e : 124 >>>	124 byte
m6 : 30460 >>>	29.7 Kb
	7 : 87760 >>>	85.7 Kb
⸮@l⸮V : 18271 >>>	17.8 Kb
<o⸮⸮ : 124 >>>	124 byte
0= : 37932 >>>	37.0 Kb

<eڋ : 62379 >>>	60.9 Kb
m: : 46356 >>>	45.3 Kb
A?y⸮ : 124 >>>	124 byte
y3 : 37334 >>>	36.5 Kb

gioreva, 2019-12-30:
Set 8 000 000
Measured 8 Mhz

Wrong Read

************************
COMMAND7: List directory
************************

7@rqI : 27318 >>>	26.7 Kb

<e : 124 >>>	124 byte
m6 : 30460 >>>	29.7 Kb
	7 : 87760 >>>	85.7 Kb
⸮@l⸮V : 18271 >>>	17.8 Kb
<o⸮⸮ : 124 >>>	124 byte
0= : 37932 >>>	37.0 Kb

<eڋ : 62379 >>>	60.9 Kb
m: : 46356 >>>	45.3 Kb
A?y⸮ : 124 >>>	124 byte
y3 : 37334 >>>	36.5 Kb

gioreva, 2019-12-30:
4 Mhz

************************
COMMAND7: List directory
************************

7@rqI : 27318 >>>	26.7 Kb

<e : 124 >>>	124 byte
m6 : 30460 >>>	29.7 Kb
	7 : 87760 >>>	85.7 Kb
⸮@l⸮V : 18271 >>>	17.8 Kb
<o⸮⸮ : 124 >>>	124 byte
0= : 37932 >>>	37.0 Kb

<eڋ : 62379 >>>	60.9 Kb
m: : 46356 >>>	45.3 Kb
A?y⸮ : 124 >>>	124 byte
y3 : 37334 >>>	36.5 Kb

gioreva, 2019-12-30:
FAT or ExFat crash CPU
FAT32 work well also at 12Mhz
Write loop 10000 time, result OK
The limit 12Mhz is your ?
I think Compiler has limit 25Mhz

************************
COMMAND7: List directory
************************

SYSTEM~1    : 0 >>>	0 byte
ACDC    WAV : 7029374 >>>	6.7 Mb
TEST1   TXT : 97000 >>>	94.7 Kb

djuseeq, 2019-12-31:
No, the limit of the SPI clock is up to the MCU and the oscillator frequency, e.g. in Atmega328 the maximum SPI clock rate is OscillatorFreq/2, on Arduino with 16MHz clock rate/2 = 8MHz.

gioreva, 2019-12-31:
SAMD has clocked at 48Mhz

La SPI dovrebbe arrivare a 20 Mhz

I worry that sometimes crash the CPU and / or the PC
I do not understand the cause.
There are a way to check if Flash is formatted properly before reading it to write it ?

https://www.arduino.cc/en/Reference/SPISettings
https://forum.arduino.cc/index.php?topic=353678.0

djuseeq, 2019-12-31:
See the example code and/or api reference, try with `getFileSystem()`

djuseeq, 2019-12-31:
To check the stability , please try to use with Interrupt pin and constructor and try different thumb drives.Check [this](https://github.com/djuseeq/Ch376msc/issues/7) issue, @Fenix-125 already tried to format the flashdrive with Win10 without success.Try  to format with some flash formatting tool or with different OS.The CH376 chip is choosy, but we can't change that.

djuseeq, 2020-01-05:
> TFT now use 12Mhz SPI
> The waveforms are very rounded, I do not think can work at 24Mhz on My PCB.
> They are very long tracks, I have SPI 6 items on PCB.
> 
> You can add the SD commands ?
> They are the same TFT
> I think they're already in the Arduino IDE
> 
> SD.print
> SD.println
> ecc.

Hi ,it support SD card now, you can [test](https://github.com/djuseeq/Ch376msc/tree/SDcard) it. If you want to try with ebay module, be aware probably on your board is made the same mistake, the SD card's CS pin (23) is connected to GND! I will write some info how to fix it

gioreva, 2020-01-05:
My board has SD directly on the SPI.
I halso have special booloader that upgrade firmware using SD

I meant to implement the commands print, library already present on Arduino IDE

About Flash formatting, Win10 has 3 kind of FAT
ExFAT
FAT
FAT32

I have used 1GB SD on SD to USB adapter.
One or two of this kind work well.

gioreva, 2020-01-05:
This
https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/Print.cpp

gioreva, 2020-01-05:
This command:
https://www.arduino.cc/en/reference/SD

djuseeq, 2020-01-05:
Ok, now i see. Sorry but i'm not planning to implement Print class in this library. I think it would increase the size of the code especially if you had to use the File class as well(maybe think wrong because I don't have much experience with it) and if the Zero works well then this issue can be close and open a new one if you find a problem.

closed 2020-01-07 by djuseeq