djuseeq/Ch376msc #12: Append to new file

andrewerrington, 2019-12-06:
I thought I had broken things when I decided that appending to a new file was the same as appending to an existing file. It seems to me that it should be. However, if I try to do this I end up with a corrupt filename. You can reproduce this by running the test code and selecting option 2:Append _without_ first selecting option 1:Create.

I am writing a log file to the USB drive. I want to append continuously to the log file, but I need to handle the case where the drive is blank at first, or if someone removes the USB drive with the log file on it and inserts a new one with no file on it. This means that every time it is time to write to the log I must first check if the file exists, then create it and write to it if it doesn't exist, or move the cursor to the end and write to it if it does exist. This is cumbersome, but then, so is everything in the programming world.

What I want to do is openFile(), moveCursor(CURSOREND), then writeFile() every time. If the file is new it will be created and will be empty, and moveCursor() will essentially do nothing. Is this possible? I hope so.

Thanks for the great library!

djuseeq, 2019-12-06:
Hi, yes i reproduced. Actually `flashDrive.openFile();` function returns a value.If the file is not exist then return 43h what is ERR_FOUND_NAME answer from the chip. So it has to be "if condition". `if(flashDrive.openFile() == ANSW_USB_INT_SUCCESS){             
        flashDrive.moveCursor(CURSOREND); 
        }` .Thanks, i will update the example code.

andrewerrington, 2019-12-07:
Thanks! My current workaround is openFile() then getFileSize(). If the file length is zero then I can use writeFile(), but if it's not then I use moveCursor() then writeFile().

I'm using your code for a simple data logger. I have noticed a couple of odd things, so if I can't get them working then I'll post an issue here. Thanks for taking a look at this one.

djuseeq, 2019-12-07:
Ok, I'm glad I could help. Then i will close this issue and if you find more problems,then you are welcome to open a new issue ;)

closed 2019-12-07 by djuseeq