djuseeq/Ch376msc #13: HID Lib to communicate with input peripherals: mouses, keyboards, gamepads, tablets, etc.

rtek1000, 2019-12-06:
Hello, I saw that it is possible to use CH376 with HID devices, could anyone indicate any library?

I would like to use a keyboard.

I found these references:

http://en.pudn.com/Download/item/id/2056626.html

> Description:   CH376 operation of the mouse keyboard, can bus, device descriptor configuration descriptor, mouse, normal operation. 

http://pulko.mandy.pagesperso-orange.fr/shinra/albireo.shtml

> Practical case: driver for an HID mouse
> 
> Since the early developments on Albireo, things have changed a little in the CPC world and several other mass storage interfaces are now easily available. However, another problem was still not completely solved: there was not a simple solution for connecting a mouse to the CPC.
> 
> Of course, the Albireo USB port can drive any USB peripheral, including any standard USB mouse. It is no more needed to dig out a PS/2 compatible mouse for your Symbiface, or even worse, an Atari/Amiga one for the MutliPlay.
> 
> Moreover, the USB standard provides a rather high level view on events from the mouse, which makes implementing this with low CPU use possible (the mouse driver was initially written completely in BASIC, and if using assembler, an interrupt driven setup is possible, freeing most CPU cycles).
> 
> This also serves as an example of how USB communications are implemented with the CH376. It opens the possibility of writing drivers for more devices. Gamepads, printers, webcams... you name it.
> 
> The code in this section is written in BASIC-like syntax. This makes it easy to read for everyone.

djuseeq, 2019-12-07:
Hi, I've written a HID version of lib from my msc library. But I didn't have time to finish. I need to dive deeper into the USB protocol. The mouse works, but the keyboard not. Maybe in the future.

rtek1000, 2019-12-07:
Ok, thanks, I'm trying to build a HMI datalogger and I need a USB keyboard, and USB stick to work with an ILI9341 display.

I found a library to operate with the MAX3421E, which is a dedicated USB Host chip, but couldn't make the library work with STM32. This library works with HID and other functions. (Lib.: https://github.com/felis/UHS30) 

Interesting that there are microcontrollers with two USB Host ports, but libraries are not yet available to operate with HID (Keyboard) and MSC (Pendrive).

I believe libraries will soon be available to operate USB Host / OTG on a variety of chips, but I think possibly they will be available at the same time. 

I also found several usage examples available since 2015, but the compiler is very expensive:

> Features
> 
>     Use USB on STM32Fxxx devices
>     Supports USB FS and HS ports
>         HS mode can use external PHY or embedded phy featuring USB HS in FS mode
>     Supports both USB modes at the same time
>     Supports different HOST or DEVICE modes at a time
>     Supports changing USB mode (host/device) on the fly on single USB FS or HS mode
>     Built on ST’s stack for USB Host/Device
>     Predefined support for different STM32F4/7xx Discovery boards for easily use
>     Supports different USB classes for host mode
>         MSC host for operating with USB flash memory
>         HID for reading external keyboard and mouse
>     Supports different USB classes for device mode
>         CDC device for acting like Virtual COM Port (VCP) to your computer
>             Requires ST’s Virtual COM port drivers
>         MSC device for acting like SDCARD reader when SDCARD is connected to SDIO port
>     Supports one USB class on both USB FS or HS ports at a time if needed
>         Example MSC: You can connect 2 flash drives at a time to make copy/paste files between them
>         Example HID: Read keyboard and mouse at the same time on both USB ports
> 

http://stm32f4-discovery.net/2015/08/hal-library-21-multi-purpose-usb-library-for-stm32fxxx/

closed 2019-12-07 by rtek1000