djuseeq/Ch376msc #15: Force USB presence detection

andrewerrington, 2019-12-08:
I am experiencing inconsistent behaviour from .checkDrive().

I am building an Arduino data logger which has three softserial ports. One of these is for the CH376. The other two are for devices that give me data. The built-in (hardware) serial port is hooked up to a PC (via USB) for debugging.

If I call .checkDrive() at startup I can reliably detect that a memory stick is or is not present. If I call it again later then it doesn't reliably report that the memory stick is or is not present. I wonder if it's because I am switching to the other softserial ports (using .listen()) in between using the CH376 softserial port? I always use .listen() to select the CH376 softserial port before doing any CH376 operations.

Is there a way to force the Ch376msc object to start from scratch? Or a way to reset its status so I can call .checkDrive() again?

Before writing the log file I call .checkDrive(). If I start my data logger with a memory stick inserted then I can write files reliably. But, if I take out the memory stick the program will still try to write files because .checkDrive() returns true.

If I start my data logger without a memory stick inserted then I can't write files because .checkDrive() returns false . But, if I insert the memory stick the program will still not write files because .checkDrive() still returns false.

Your test program works properly. I can insert and remove the memory stick and the test software will detect the current state. When your test program is running or my program is running the indicator LED on the CH376 board turns on and off correctly when the memory stick is installed or removed. But, when I call .checkDrive() in my program it doesn't report the current state properly. What am I missing? Do you have any suggestions for debugging? Thanks.

djuseeq, 2019-12-08:
Hi, the SoftSerial have some [limitations](https://www.arduino.cc/en/Reference/SoftwareSerial). If the module is initialized and UART communication is in use then when an external event occurs(e.g. drive detach/ drive attach) the chip will send immediately a interrupt message to the Arduino. So if the controller fetching data from an another SoftSerial port and in this time the USB chip send a message, then this message will be probably lost. Before file operation try to call .mount() function. If the return value is 14h(ANSW_USB_INT_SUCCESS) then there is a device attached to the module, if 82h(ANSW_ERR_DISK_DISCON) then no device is attached.

andrewerrington, 2019-12-09:
Ok. I will try that. But I might also re-work my design and use SPI for the CH376 instead of softserial.

Thanks!

djuseeq, 2019-12-09:
Ok. I think that is more reliable than SoftSerial. You can try [this](https://github.com/djuseeq/Ch376msc/tree/SAMD) branch with updated library. If everything will work properly (i have to test is also) then i will update the MASTER branch.

closed 2020-01-07 by djuseeq