djuseeq/Ch376msc #18: ESP32: Attach flash/SD drive first! - how can I help debug (:

petervanderwalt, 2020-01-17:
I have an CH376S hooked up to an ESP32 via SPI. 
Example sketches compile fine, but cannot read from USB drive: "Attach flash/SD drive first!"

If there is anything I can do to help you get it supported on ESP32 (or am I just being stupid, I have the correct CS pin wired/assigned)

Schematic (own board)

djuseeq, 2020-01-18:
Hi, that would be great to know if the library works with ESP32. If the ACT LED turns on / off when you connect / disconnect the drive, that means the chip is initialized, which is good. You will receive an "Attach ..." message if no drive is connected to the Host or there is probably a problem with the INT pin (14). If you can, try a different pin, or try to hookup  LED + resistor to this pin and then toggle it.

djuseeq, 2020-01-18:
Or in a "worst" case, because i like gadgets :) i will order one ESP32 CAM board and then i can try with this lib and at least get to know the device better. But this takes time, anyway i think I convinced myself. ;)

petervanderwalt, 2020-01-18:
That would be awesome!

As its our own board, it could be that too, so hard to say, maybe its the
PCB (99.9% sure the circuit is correct, straight out of datasheet, with all
the notes about INT and pullups, mode settings, etc applied) when I dont
know for sure it should work on the ESP32 (:

If you get yours and confirm it does, that would be a great help going
forward - although i am hoping its something in the library, there are
little differences between esp8266 and esp32 that i tripped over on other
libraries too

petervanderwalt, 2020-01-20:
Did a little more digging today:

- I could see the INT pin going low when I plug/unplug drive, so I knew the chip was working great... (but no response to checkIntMessage telling me comms issue somewhere)
- Although I am sharing the bus with other devices, I didn't think that would be the problem (as I am setting it with the INT pin?) but as soon as I desoldered the XPT2046 touchscreen controller IC from my board, the CH376S started working!   
So yay, your library works on ESP32, and also Yay, my board works. 
But how can I share the SPI bus now? 

```
//If the SPI port shared with other devices e.g SD card, display, etc. remove from comment the code below and put the code above in a comment
Ch376msc flashDrive(4, 14); // chipSelect, interrupt pin

```

Note1:  In testing this I only use your provided basicUsageSPI test, the other peripherals are not even being initialised, none of the libraries loaded, etc
Note2: I plan on sharing the SPI bus with the CH376s, with a XPT2046 Touch controller and ILI9341 TFT LCD

petervanderwalt, 2020-01-20:
OK further testing, I can share the bus!   SD card on same SPI bus, inits, and CH376s still works!  So its just someting with the XPT2046

I think we can close this one out (and you can add ESP32 working to the Readme - didnt need any tricks)  - I'll go check the XPT2046 datasheet to see why it interferes instead (:

closed 2020-01-20 by petervanderwalt

djuseeq, 2020-01-21:
Thanks, i will add it, please if you found the culprit for XPT2046 freezing then let i know. ;) I just had a quick idea, when you try the example code and there is more SPI devices on the bus then i recommend to put all unused devices ChipSelect pin's to HIGH otherwise if they lack's of pullup resistor's then these lines are floating, they can accidentally get enabled state and ruins the communication.

petervanderwalt, 2020-01-21:
If I initialise the XPT2046 before the CH376 is works perfectly. 

So you may be onto something, this board doesn't have pullups for CS on every device, and it is actually a good idea, will add it on the next prototype run! 

Later today I will try again with just setting the XPT2046's CS pin High and see if it still interferes

djuseeq, 2020-01-21:
Nice to hear it works. Glad I could help.

microElabDevelop, 2022-01-30:
@petervanderwalt hi u run ch376  module with esp32. So kindly tell me this lib work with esp32 spi mode proper  or u made change . Kindle give me example . Thankz

petervanderwalt, 2022-01-30:
> @petervanderwalt hi u run ch376 module with esp32. So kindly tell me this lib work with esp32 spi mode proper or u made change . Kindle give me example . Thankz

We use the library as-is and it works great. If you have issues its almost always your hardware

microElabDevelop, 2022-01-30:
@petervanderwalt  thankz for response 
I use this setting

Tx  > GND
WR > GND
RD > GND
CS > 3V3
D7 > pin_spi_miso  = esp32 ->D19
D6 > pin_spi_mosi  = esp32 ->D23
D5 > pin_spi_sck     = esp32 ->D18
D3 > pin_ch376_cs = esp32 ->D5
INT > pin_ch376_int =NOT USE bcz no other device i use on spi.
Ch376msc flashDrive(5);
What anything i miss.?

microElabDevelop, 2022-01-31:
thanx for responce , ur right , TX pin remove from GND and problem solve, im little bit confuse in SPI and parallel port. again thanx

COYOO-COYOO, 2025-08-15:
> 今天做了一些进一步的挖掘：
> 
> * 当我插入/拔出驱动器时，我可以看到 INT 引脚变低，所以我知道芯片工作正常......（但没有对 checkIntMessage 做出响应，告诉我某处存在通信问题）
> * 虽然我和其他设备共享总线，但我没想到这会是个问题（因为我用 INT 引脚设置了它？）。结果，当我从电路板上拆下 XPT2046 触摸屏控制器 IC 后，CH376S 立刻就开始工作了！
>   所以，你的库可以在 ESP32 上运行，我的电路板也能正常工作了。
>   但是，我现在该如何共享 SPI 总线呢？
> 
> ```
> //If the SPI port shared with other devices e.g SD card, display, etc. remove from comment the code below and put the code above in a comment
> Ch376msc flashDrive(4, 14); // chipSelect, interrupt pin
> ```
> 
> 注意 1：在测试中，我仅使用您提供的 basicUsageSPI 测试，其他外设甚至没有初始化，没有加载任何库等。 注意 2：我计划与 CH376s 共享 SPI 总线，使用 XPT2046 触摸控制器和 ILI9341 TFT LCD

Hi, can you provide an example compatible with ESP32? I have been trying to debug but I can't seem to initialize CH376.