djuseeq/Ch376msc #20: Add Test for chip presence

gioreva, 2020-01-20:
Add Test for chip presence

And what difference from driveReady() and checkDrive()  ?
Always boot true if flash in, and boot false in the absence of flash

```
  if(flashDrive.init()) SerialUSB.println("Chip Ok");      
  else SerialUSB.println("Chip fail");
  
  if(flashDrive.driveReady()) SerialUSB.println("driveReady Ok");      
  else SerialUSB.println("driveReady fail");
   
  if(flashDrive.checkDrive()) SerialUSB.println("checkDrive Ok");      
  else SerialUSB.println("checkDrive fail");

/////////////////////////////////////////////////////////////////
bool Ch376msc::init(){
.....	return _controllerReady;
}
```

djuseeq, 2020-01-20:
Hi, in the [1.4.2](https://github.com/djuseeq/Ch376msc#versions) version i have changed the name because the checkDrive function name was confusing. With ~~checkDrive~~ `checkIntMessage()` function basically we polling the chip's interrupt pin,not necessary but sometime it can be useful.The chip trigger an interrupt if usb device is connected/disconnected. The  `driveReady()` function query the chip's device status(usb drive are connected,ready or not). Please check the updated library's example files. You can [check](https://github.com/djuseeq/Ch376msc#api-reference) the reference.

djuseeq, 2020-01-20:
With [`pingDevice()`](https://github.com/djuseeq/Ch376msc#api-reference) function it can test the chip's presence.

closed 2020-06-07 by djuseeq