djuseeq/Ch376msc #22: Program not respond in use an unknown partition

djuseeq, 2020-01-23:
But this is a problem.
If the type of key or formatting is not recognized, you can not completely stop the program.

_Originally posted by @gioreva in https://github.com/djuseeq/Ch376msc/issues/19#issuecomment-577609978_

djuseeq, 2020-01-23:
@gioreva Please test this test [release](https://github.com/djuseeq/Ch376msc/tree/bug%2322)

gioreva, 2020-01-24:
better but not enough.
Before the SD to USB adapter, no SD inserted, blocked the program.
Now he is not blocked, however, SD 128GB NTFS, has halted

gioreva, 2020-01-24:
I have blinking led on the loop() for check this

djuseeq, 2020-01-24:
Sorry but maybe i don't understand. What is SD to USB adapter? SD [library](https://www.arduino.cc/en/Reference/SD) for SD card supports FAT filesystem, NTFS is not supported.

gioreva, 2020-01-24:
This is SD adapter [link](https://www.ebay.it/itm/BG-Super-Speed-USB-2-0-Micro-SD-SDXC-TF-Card-Reader-Adapter-for-Mac-OS-System-U/293136277123?hash=item4440489e83:g:VNgAAOSw1AZceKv-)

Many device are not supported, but can not block the program and all the board functions.
Whatever answer is accepted, but not to kill the card.

djuseeq, 2020-01-24:
Ahha, USB card reader, i don't have experience with card reader handling on this module. But this chip can handle only FAT formatted drives. The card is destroyed? Sad to hear that , PC recognize it? Or problem is not with the card reader? Block the program? Have you [read](https://github.com/djuseeq/Ch376msc/issues/18#issuecomment-576648374) this answer? You have almost the same hardware setup, many devices on the SPI bus and if their CS pin is not pulled to logical HIGH state then it would ruins the communication and you will face that kind of problems. If you are not sure then in the beginning setup() function configure all the CS pins (for EVERY devices) Output and put them on HIGH state or use pullup resistors otherwise those pins are floating.

gioreva, 2020-01-24:
the problem is that your software hangs forever if the file system is ntfs. In this case, it should abort read and keep running, not just hang forever! I understand I should use fat. This is not a problem, I use fat. But if by mistake I use ntfs, the entire program hang or crash. You should handle an error condition or just return an error and not stop/crash.

djuseeq, 2020-01-24:
My program is not perfect, i writed for myself because i didn't found a usable library/code for this module. I'm not a professional programmer, this is just my hobby. I started to learn coding 5 years ago and mostly i learned from communities so when i finished this code i had that feeling to if i make it available to others then i give back something to community. Yes i'm really happy if i get feedback (including bugs) because then i have possibility to fix it, improve it. Please read [this](https://github.com/djuseeq/Ch376msc#license). I really want to help/fix the bug but please be patient, it's really difficult if we don't understand each other and i coding in my free time. Im working at night i have family so i don't have so much free time. No one is forced to use this code if it's so bad and maybe is better if i don't bother with developing, slowly i don't enjoy it anymore. Sorry for this essay, maybe again is just a misunderstood.

djuseeq, 2020-01-24:
Now i tried with [bug22](https://github.com/djuseeq/Ch376msc/tree/bug%2322) branch, and doesn't "hangs forever", for NTFS filesystem the Chip send 0xA1 answer (partition error)
[image]

gioreva, 2020-01-24:
Do not worry, I was not criticizing your great and appreciated work! I understand you, and I don't think I would ever been able to do the same work you did.
This is the [code](https://1drv.ms/u/s!AoR_HuKwnabrwDP90XH2czS7vcrj?e=ZARMo0)
It is you example SPI
I send 7 for file list.

This monitor.
Plugghin many type of flash key.
When i plug 128GB NTFS, immediatly led stop blynking, no message appairs, and no other command are acceptet, until reset.

[

*****************************************
h:Print this help

1:Create
2:Append
3:Read
4:Read date/time
5:Modify date/time
6:Delete
7:List dir
8:Print free space
9:Open/Create folder(s)/subfolder(s)
*****************************************

Flash drive attached!

************************
COMMAND7: List directory
************************

/LOST    DIR
/TWRP
/APK
/ANDROID
/LOST~1  DIR
FOUND   000 : 0 >>>     0 byte
RECOVE~1IMG : 13199360 >>>      12.6 Mb
RECOVE~2IMG : 13172736 >>>      12.6 Mb
/CLIENTI
RE5751~1IMG : 14125056 >>>      13.5 Mb
RECOVE~4IMG : 13189120 >>>      12.6 Mb
RECOVE~3IMG : 13199360 >>>      12.6 Mb
TWRP_3~1IMG : 6037376 >>>       5.8 Mb
BOOTEX  LOG : 42716 >>> 41.7 Kb
SYSTEM~1    : 0 >>>     0 byte
MODULO~1PDF : 86674 >>> 84.6 Kb
RECOVERYIMG : 13172736 >>>      12.6 Mb
CLIENT~2RAR : 176827283 >>>     168.6 Mb
CLIENT~1RAR : 1289785 >>>       1.2 Mb
CLIENT~3RAR : 176827107 >>>     168.6 Mb
CLIENTI RAR : 1878991 >>>       1.8 Mb
IT      RAR : 5429334 >>>       5.2 Mb

*****
Done!
*****

Flash drive detached!
Flash drive detached!

************************
COMMAND7: List directory
************************

*****
Done!
*****

Flash drive detached!
Flash drive detached!

************************
COMMAND7: List directory
************************

*****
Done!
*****

Flash drive detached!
Flash drive attached!

************************
COMMAND7: List directory
************************

SYSTEM~1    : 0 >>>     0 byte
TEST1   TXT : 97000 >>> 94.7 Kb
CONFIG  INI : 2162 >>>  2.1 Kb
EXAMPLE TXT : 0 >>>     0 byte
/BACK
ACDC    WAV : 7029374 >>>       6.7 Mb
TESTO   WAV : 451278 >>>        440.7 Kb
1       WAV : 48440 >>> 47.3 Kb
2       WAV : 50120 >>> 48.9 Kb

*****
Done!
*****

Flash drive detached!](url)

djuseeq, 2020-01-27:
Hi, please download the updated library from link given above and test it again.If still there is issue then please let  me know about it. Main problem is there is lack's of documentation so many things i have to find it out with lot of testing and analyzing with logic analyzer. Thanks

djuseeq, 2020-01-27:
I don't have flash drive with this high capacity so i can't test with it.

gioreva, 2020-01-27:
Ok, 
now do not break at flash insertion, but break at 7 command.
List dir.

djuseeq, 2020-01-27:
It's happening just with 128Gb NTFS formatted flash drive or with other
devices also?

gioreva, 2020-01-28:
I tried 10 kinds of keys.
When I insert this, all break just inserted it.
16GB, fat32
But the problem, in my opinion, is that there must be a timeout if the chip is not responding, with subsequent reset.

[image]

Othert brand of fat32 16GB, work well.

djuseeq, 2020-01-28:
Ok, timeout is already implemented where it can stuck in an endless loop, i think it can stuck in a while loop in example file. I modified the lib, please if you can download it again and if still stuck then please try to edit the example code: (4. line)
`      case 55: //7

        printInfo("COMMAND7: List directory");

          while(flashDrive.listDir()){ // reading next file

            Serial.println(flashDrive.getError(),HEX); // error code or 0 if no error

            if(flashDrive.getFileAttrb() == ATTR_DIRECTORY){//directory

              Serial.print('/');

              Serial.println(flashDrive.getFileName()); // get the actual file name

            } else {

              Serial.print(flashDrive.getFileName()); // get the actual file name

              Serial.print(" : ");

              Serial.print(flashDrive.getFileSize()); // get the actual file size in bytes

              Serial.print(" >>>\t");

              Serial.println(flashDrive.getFileSizeStr()); // get the actual file size in formatted string

            }
          }

          printInfo("Done!");

        break;`

gioreva, 2020-01-28:
New lib, old .ino
Same problem.
One key break at the insertion and another to 7 command.

Your change do not compile.

exit status 1
'class Ch376msc' has no member named 'getError'; did you mean 'getYear'

djuseeq, 2020-01-28:
Weird, if there is no getError function then probably the library is not updated to [this](https://github.com/djuseeq/Ch376msc/tree/bug%2322) version. You have to download it and manually update( overwrite the files in Ch376msc directory )

gioreva, 2020-01-28:
Opps, sorry, i donload official lib and not bug.
Now I tried with that bug

With boot key, at insertion do not crash.
Timeous is one second ? Too much.
Can be 0.1 sec ?

But booth, at 7 command crash

djuseeq, 2020-01-28:
You can reduce the timeout in Ch376msc.h file, but sometime the chip is busy and then longer timeout required, so if you reduce it to too low then problem can happens. Did you get back some error code whit this line ` Serial.println(flashDrive.getError(),HEX); // error code or 0 if no error` when it stuck?

gioreva, 2020-01-28:
If that line is already active, I do not receive any error.
My sketh is old, except for the changes that you require.

```
*****************************************
h:Print this help

1:Create
2:Append
3:Read
4:Read date/time
5:Modify date/time
6:Delete
7:List dir
8:Print free space
9:Open/Create folder(s)/subfolder(s)
*****************************************

Flash drive attached!

************************
COMMAND7: List directory
************************

```

gioreva, 2020-01-28:
But the serial dies after the last line.

djuseeq, 2020-01-28:
Ok, i have to check it, because i can't reproduce it. The problematic device is simple flashdrive or usb cardreader on link what you shared before?

djuseeq, 2020-01-28:
Please try to update and test it again :(

gioreva, 2020-01-28:
One is normal 16GB fat32, other is adapter (perhaps worse)
Same downloading new bug version.
Led stop blinking for one second at insertion, then stop definitively to the 7 command

`

*****************************************
h:Print this help

1:Create
2:Append
3:Read
4:Read date/time
5:Modify date/time
6:Delete
7:List dir
8:Print free space
9:Open/Create folder(s)/subfolder(s)
*****************************************

Flash drive attached!

************************
COMMAND7: List directory
************************
`

djuseeq, 2020-02-03:
Hi, i have updated and uploaded again but i can't reproduce this error. If you have CH376 on module also then try with different architecture(e.g.8bit AVR, Uno board), the same problem occurs? Or if you have it just on your PCB then try to do not use other libraries, just in the very beginning put all devices on SPI bus in inactive state(CS pins). Do you have logic analyzer? I am very curious about the communication on the SPI bus when the error occurs. Regards

djuseeq, 2020-02-03:
Or one more idea, try on UART

gioreva, 2020-02-03:
I would like to help by analyzing the code.
The problem is that they are late with all my projects.
I also decided that I mount optional CH376, on my card, because it is expensive.
You can add as many debug print?
So I am sending you a result to understand where and why crashes.
When I have time I will try to analyze the code.

gioreva, 2020-02-03:
However, if you have made entered the code for timeout insertion, only mistakes to not accept other commands after this.

gioreva, 2020-03-07:
Forcing higher the interrupt pin, the program crashes.

djuseeq, 2020-03-07:
Hi, Int pin does not need pull-up resistor, put the pull-up resistors to the ChipSelect pins, and you can try [this](https://github.com/djuseeq/Ch376msc/tree/update) version. If that doesn't help, then if I have some free time i will write a debug version to see where it stuck.

gioreva, 2020-03-09:
Same using this download.
I have no resistor on INT pin, but when the chip is not welded is floating
I could put a resistor to ground, but now the card is in production.
The test program prints nothing, stops at init(), if INT is high.

  flashDrive.init();
  delay(100);
  tft.println("Ch376 test");
  if (flashDrive.pingDevice())     tft.println("Ch376 ok");
  else   tft.println("Ch376 fail");

gioreva, 2020-03-09:
Opps, sorry, crash is forcing INT LOW