djuseeq/Ch376msc #23: writefile(string)

thquack, 2020-01-29:
I am trying to use the USB as a csv data logger and I am not sure if I am off, but char writing seems a bit cumbersome, vs #variables particular with #number values such as int.

is it possible to directly write an integer?

The ch376msc guide states:
`flashDrive.writefile(string, strlen(string)); //string, string length`  
//but string really this is really a **char** string

for data logging it is fairly straight forward to add int, floats, etc.

//standard sdFat.h then example is
file.print(float); file.println();

//CH376msc.h requires example
string.toCharArray(charstring,10)
//then 
writefile(charstring, strlen(charstring));
`
this all gets quite cumbersome when dealing with multiple data variables including floating points.

djuseeq, 2020-01-29:
Hi, this lib can read and write char type of variables(c type string alias char array), so actually yes is possible converting to integer or vice versa just need to write these routines which make the conversions.  If there is a need I can think about it and maybe write these functions.

djuseeq, 2020-02-03:
Hi, if you can, please download and test [this](https://github.com/djuseeq/Ch376msc/tree/bug%2322) version of the library, some new functions and example are available, e.g. writing and reading integer/float numbers. But testing is required.

thquack, 2020-02-03:
thanks!
I have downloaded the example.  It compiles, but as a side note, it uses a lot of program storage space memory.  
I will work to incorporate it into my data-logger and test it live...results to follow.

thquack, 2020-02-04:
Ok, I integrated this into a generic test for a temperature controller data logger.  The update works.  Is it possible to /newline after a char string?
`#include <SPI.h>            //SPI bus
#include <Ch376msc.h>         //USBFLASH 
#include <SoftwareSerial.h>   //serial debug

//pins
#define USBFLASH_CS   5
#define USBFLASH_INT  8

//---------------------------------------------------------------------
//USBFLASH 
Ch376msc flashDrive(USBFLASH_CS, USBFLASH_INT); // chipSelect, interrupt pin
boolean usbError = false;

//filename creation
#define USB_FILE_BASE_NAME "FDdata"  //base file name up to 6 characters
const uint8_t BASE_NAME_SIZE = sizeof(USB_FILE_BASE_NAME) - 1; 
char USBfileName[13] = USB_FILE_BASE_NAME "00.csv"; //logfile naming

// buffer for reading
char adatBuffer[255];// max length 255 = 254 char + 1 NULL character
boolean readMore;

char headerString[] = "Time(ms),record,Set_Temp,Mode,Temp_Read,PID_percent,P,I,D\0"; //the csv header
char commaString[] = ","; //a comma for CSV

//Sample temperature controller variables
char mode[8] = "Auto";  //mode text for log
float temp_read = 60.0; //the tempature reading
byte set_temp = 153;  //the set tempature
float PID_value = 0; //PID Duty in % 
volatile static uint32_t elapsedTime, timePrev, logTime, Time = 0, now = 0, last_time, elapsedTime_log, timePrev_log; //real time clock
//output values
float PID_p = 0;    float PID_i = 0;    float PID_d = 0;

int dd = 0;  //log record counter

word Ts;  //PID update sampletime in sec (user inputs)
word loop_rate; //main loop rate
word log_data_now;  //lof file sample rate

//---------------------------------------------------------------------------------------------
void setup() {
Serial.begin(57600);  //Serial.println("hello, Serial is up");
//USBflash
flashDrive.init();
if(flashDrive.getDeviceStatus()){
     usbError=false;
    } else {
      usbError=true;
      Serial.print("USB ERROR at Init");
      exit(0);
    }
checkUSB();

now=millis(); // read RTC initial value
elapsedTime = now; // prepare for next RTC delay loop 
loop_rate = 200;  // 5Hz main loop (ms)
Ts = 5; //5 seconds PID loop (sec)
log_data_now = 1;//log to USB sample time (sec)

Serial.print("Starting at File:  ");
Serial.println(USBfileName);
flashDrive.setFileName(USBfileName);

//log to a unique file (add 1 and look)
while (flashDrive.openFile() == ANSW_USB_INT_SUCCESS)   {
    Serial.print("Found File:  ");
    Serial.println(USBfileName);
    if (USBfileName[BASE_NAME_SIZE + 1] != '9') {
[... 168 lines cut ...]
Found File:  FDdata33.csv
Found File:  FDdata34.csv
Found File:  FDdata35.csv
Here we go, writing to:  FDdata36.csv
Write header...
in header, writing: Time(ms),record,Set_Temp,Mode,Temp_Read,PID_percent,P,I,D
Header written: FDdata36.csv
log: 1, log: 2, log: 3, log: 4, log: 5, log: 6, log: 7, log: 8, log: 9, log: 10, log: 11, log: 12, log: 13, log: 14, log: 15, log: 16, log: 17, log: 18, log: 19, log: 20, log: 21, log: 22, log: 23, log: 24, log: 25, log: 26, log: 27, log: 28, log: 29, log: 30, log: 31, log: 32, log: 33, log: 34, log: 35, log: 36, log: 37, log: 38, log: 39, log: 40, print file: FDdata36.csv
Time(ms),record,Set_Temp,Mode,Temp_Read,PID_percent,P,I,D0
1058,1,153,Auto,138.23,0.00,0.00,0.00,0.00
2058,2,153,Auto,103.67,0.00,0.00,0.00,0.00
3059,3,153,Auto,135.09,0.00,0.00,0.00,0.00
4059,4,153,Auto,119.38,0.00,0.00,0.00,0.00
5059,5,153,Auto,10
6.81,57.00,-10440.00,-9067.00,-6901.00
6059,6,153,Auto,153.94,57.00,-10440.00,-9067.00,-6901.00
7059,7,153,Auto,163.36,57.00,-10440.00,-9067.00,-6901.00
8059,8,153,Auto,103.67,57.00,-10440.00,-9067.00,-6901.00
9059,9,153,Auto,116.24,57.00,-10440.00,-
9067.00,-6901.00
10059,10,153,Auto,150.80,8.00,17944.00,15668.00,-8510.00
11059,11,153,Auto,141.37,8.00,17944.00,15668.00,-8510.00
12059,12,153,Auto,125.66,8.00,17944.00,15668.00,-8510.00
13059,13,153,Auto,109.96,8.00,17944.00,15668.00,-8510.00
...`

djuseeq, 2020-02-04:
It's possible if you add `\n` character at the end of the string. I can rewrite the `writeNum` function to have a conditional terminator character, e.g if you call just `writeNum(number)` then simply write a given number to drive, or you can call `writeNum(number, terminatorChar)` and concatenate the given character with the number.

djuseeq, 2020-02-05:
I have added function `writeChar(character)` to test lib.

thquack, 2020-02-05:
adding the \n to the end solved the newline for me, such as:
`char csv_single_row[] =  "some,csv,header,row,text\n";`

> some,csv,header,row,text
> 1,2,3,4,5

```
The writeChar works as well, such as:
flashDrive.writeNum(value1);  //some
flashDrive.writeChar(','); // a cs
...
flashDrive.writeNumln(value2); //text
```

Data logger is doing what I need

thquack, 2020-02-05:
`flashDrive.writeChar('\n');  `

works for me

djuseeq, 2020-02-05:
UR welcome. Thanks for feedback. Then we can close this issue.

closed 2020-02-05 by thquack

Espectrobr, 2024-03-25:
It is possible to register a number with three decimal places using flashdrive.writeNum(float); ? tks