djuseeq/Ch376msc #25: MIDI

t0maszduda, 2020-03-27:
Hi 

Is it possible sent MIDI data from Ch376 via TX from Arduino to USB MIDI iNTERFACE?
[image]
 
Thanks
Tomasz

t0maszduda, 2020-03-27:
as this:
http://www.hobbytronics.co.uk/usb-host-board-v24

djuseeq, 2020-03-27:
Hi Tomasz,
sorry but with this library is not possible. This lib support just only storage controller mode.

t0maszduda, 2020-03-28:
Yes I know. The question is is it possible at all  do that without checking firmware? 

Do you can make a little library for midi
which sends Serial.begin(31250) from Arduino by tx straight to usb socket

Without any command for read/write files etc.

t0maszduda, 2020-03-28:
I found something but this is USB Host Shield and I don't know is it similar to ch376 or not
https://github.com/felis/USB_Host_Shield_2.0

djuseeq, 2020-03-28:
Yes, it's possible to use with HID devices, i had an attempt in the past to write a code but i never finished. Takes too long time, and at the moment i don't have enough time. I would like to finish and if it succeeds I will make it public. With the shield in your link is possible also if you can find a lib, for file manage the ch376 is more easier because we don't need to bother with low level USB protocol or file system, just send the proper commands to the chip so our code takes up much less space in MCU.

t0maszduda, 2020-03-30:
Thanks 

I’m waiting for MIDI library:)

Best Regards
Tom