djuseeq/Ch376msc #26: Trying to use streamFile in ESP32's "webserver" library

Havocking1992, 2020-05-06:
Hello, i am trying to stream image from USB FLASH to my web, but it seems that this library is not constructed to use with webserver library. Is there workaround for this, or it needs to be implemented? Thank you.

Device ESP32 D1 R32 + Ch376s

-- Errors in Arduino IDE while compiling--

Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/WebServer.h: In instantiation of 'size_t WebServer::streamFile(T&, const String&) [with T = Ch376msc; size_t = unsigned int]':

esp32-server-usb-test1\esp32-server-usb-test1.ino:355:45:   required from here

Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/WebServer.h:138:5: error: 'class Ch376msc' has no member named 'size'

     _streamFileCore(file.size(), file.name(), contentType);

     ^

Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/WebServer.h:138:5: error: 'class Ch376msc' has no member named 'name'

Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/WebServer.h:139:37: error: no matching function for call to 'WiFiClient::write(Ch376msc&)'

     return _currentClient.write(file);

                                     ^

In file included from 
Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFi.h:37:0,

                 from esp32-server-usb-test1\esp32-server-usb-test1.ino:3:

Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFiClient.h:55:12: note: candidate: virtual size_t WiFiClient::write(uint8_t)

     size_t write(uint8_t data);

            ^

Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFiClient.h:55:12: note:   no known conversion for argument 1 from 'Ch376msc' to 'uint8_t {aka unsigned char}'

Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFiClient.h:56:12: note: candidate: virtual size_t WiFiClient::write(const uint8_t*, size_t)

     size_t write(const uint8_t *buf, size_t size);

            ^

Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFiClient.h:56:12: note:   candidate expects 2 arguments, 1 provided

Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFiClient.h:58:12: note: candidate: size_t WiFiClient::write(Stream&)

     size_t write(Stream &stream);

            ^

Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src/WiFiClient.h:58:12: note:   no known conversion for argument 1 from 'Ch376msc' to 'Stream&'

In file included from 
Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Stream.h:26:0,

                 from 
Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:147,

                 from sketch\esp32-server-usb-test1.ino.cpp:1:

Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.h:69:12: note: candidate: size_t Print::write(const char*, size_t)

     size_t write(const char *buffer, size_t size)

            ^

Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.h:69:12: note:   candidate expects 2 arguments, 1 provided

Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.h:61:12: note: candidate: size_t Print::write(const char*)

     size_t write(const char *str)

            ^

Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.h:61:12: note:   no known conversion for argument 1 from 'Ch376msc' to 'const char*'

djuseeq, 2020-05-18:
Hi, i don't know that WebServer library, with Ch376 lib you can copy data from flash to byte or char array buffer for further use. Please can you give a link to that Library? 
Regards,
György

Havocking1992, 2020-07-12:
Nevermind, I'm not interested anymore in this, found alternative way, thank you

closed 2020-07-12 by Havocking1992