djuseeq/Ch376msc #27: Jumper position error?

trotterlg, 2020-05-20:
Is there an error in the jumper position diagram for changing between uart and spi mode?  Appears the jumper remains in either the serial or the spi position no matter which is in use.

djuseeq, 2020-05-20:
Hi, no, there is no error. Pin header i marked H1 is same but there is a difference in H2 header. With H1 you can choose between Parallel and Serial, and also here is the UART RX TX pins. This layout may vary depending on the module 
[image]

trotterlg, 2020-05-20:
Thank you for explaining this to me, makes perfect sense to me now. One thing that isn’t clear to me is how it also supports a SD card. I see that mentioned but not how it works. Your write up is by far the best I have found for using this board. We are developing a cubesat that needs to switch memory between an Arduino and a Raspberry pi. Raspberry doesn’t like additional SD cards and Arduino isn’t easy with usb. This seems perfect. We are using the Raspberry to do the pictures and processing then use the Arduino for the communications work. That allows us to shutdown the Raspberry most of the time to meet our power budget. Thanks again for the help. 

Larry Trotter
trotter@zipcon.net

djuseeq, 2020-05-20:
I'm glad I could help. Your project sounds very exciting. The chip can execute the same operations with SD card as with USB (directory listing, file read/write/delete and modify the file attributes), with a simple function call can switch between SD and USB drive, but if you own the same ebay module what i have, then some modifications is required on the board.

trotterlg, 2020-05-21:
Friday I will have my module to test and will send you a picture of it. If this works out I can eliminate one arduino and go with only one arduino and one raspberry and simplify the software lots.  I will send you a couple of pictures of the software development boards along with the usb board. Again, thanks again.

djuseeq, 2020-05-21:
Thanks , that would be awesome, I would be very happy if you could send me photos

trotterlg, 2020-05-23:
This is the board I have now, looks like yours. I do have a question about the pin marked CS which is listed as needing to be high for SPI mode, there is another pin that seems to be used for chip select that is listed as SS pin and goes to arduino pin 10 which is generally used for Chip Select with sd card readers.  Any clarification on that would be helpful. Thanks again it’s not easy finding someone who really knows about these things. 

Larry Trotter. 

Sent from my iPhone

trotterlg, 2020-05-23:
Mine is running now. I wired it as your drawing stated and it runs the test code and writes a file. On my board the pin you have marked to pull high to 3.3 is marked CS and the pin that is used for CS (marked SS on the drawing) is marked D3. Are the boards mismarked or are you using another method of device selection? 

Larry Trotter.

trotterlg, 2020-05-23:
These are the software development boards. Receiver and transmitter. I will simplify the transmitter with the usb reader. 

Sent from my iPhone

djuseeq, 2020-05-23:
These jumper configurations i found in datasheet (little-bit difficult to understand ), and these markings on the module is for 8bit parallel interface(D0-D7 and control bits). What i understood from this datasheet is when the chip powered up then the chip's fw read these pins and depending on the condition of the pins state (floating, high or low) will determine the communicating mode (parallel, UART or SPI, if UART then the communication speed also defined with these pins). When SPI is in use then these pins are factory specified (D7 - MISO, D6 - MOSI, D5 - SCK, D3 - SS( Slave Select ) or CS ( Chip Select)), on Arduino (e.g Uno) the hardware SPI port pins are(13 - SCK, 12 - MISO, 11 - MOSI) but for CS pin you can choose another pin what you like and use when you instantiate the reader object, e.g Ch376msc(YOUR_CHOSEN_CS_PIN); . Be aware, if you use another device on the SPI bus together with this module then you have to use one extra pin to receive interrupt requests from the module ( pls. check the examples). I can't see the photos, probably because you sent it as mail attachment :(

trotterlg, 2020-05-24:
I will send it full size. Thanks. Look at the CS marked pin. 

Sent from my iPhone

closed 2020-06-07 by djuseeq

BleuLlama, 2020-06-08:
I know this ticket is closed now, but I just want to mention that I have a CH367 board with a different pinout than the "standard". (this one: https://www.aliexpress.com/item/32818566592.html )  So that's fun.

djuseeq, 2020-06-08:
Hi, no problem, yes it looks like "H2" pinheader is mirrored and "H1" with jumper on is in parallel mode and without is in UART/SPI . If you figured out the correct layout and if you like to share with us then i can add it to the readme file. Thanks