djuseeq/Ch376msc #28: Windows won't list created files and folders

delgadosouza, 2020-05-31:
Dear @djuseeq , thank you for this library!

I have a BluePill (STM32F103) and also a BlackPill (STM32F401). My CH376 module is the same as yours.

For now I am using BluePill with the Arduino_Core_STM32 (by stm32duino) and it worked with no effort for the basicUsageHwSerial example! I am using HardwareSerial Serial2 on pins `PA2` and `PA3`:
```
#include <Ch376msc.h>
HardwareSerial Serial2(PA3, PA2);
Ch376msc flashDrive(Serial2, 115200);
```

The example code worked creating, appending and reading a file. It also apears to have worked creating and opening folders.

The problem is when I connect the pendrive to my computer. It appears as empty although Windows shows that some space is used on it.

I changed the example a bit and created 1000 files and was able to list the files using the module. But again, when I put the pendrive on my computer the device has no file in it, only more space is used.

I tried with an USB card reader and also with an actuall USB flash drive (pendrive).
For the USB card reader I tried formatting with FAT16 and also FAT32, for the pendrive I used only FAT32 format.
Exactly same behavior on both devices.

Any files I create in Windows also are not listed by the module. Apparently one time a file showed up in list (7) but its name was garbled.

Can you help me? Do you know what could be the problem?

djuseeq, 2020-06-01:
Hi, unfortunately I received such feedback with the same issue, please can you share more info? OS version, flash drive brand/model, size, partition type. Check this issue, @Fenix-125 already tried to use pendrive formatted with Win10 and Win XP.Try to format with some flash formatting tool or with different OS, due to lack of windows computer i couldn't test it on that OS. Maybe an OS or FlashDrive issue. It would be great to find what what cause this problem.

delgadosouza, 2020-06-01:
Hi @djuseeq , thank you for the fast answer!

I am using Windows 10 version 1909.

The SD card I was using with the USB card reader had only 2GB. I don't know how it was formatted before but when I used the default configurations for formatting it with Windows, it used the FAT format with 32K allocation unit.
When creating the 100 files (TEST_0.TXT to TEST_99.TXT) I run the command 7 to list the files and only the file TEST_0.TXT showed up. Then when I put it on my Windows PC it showed the remaining files but the TEST_0.TXT file was missing. I was able to open the files and everything was right.

Then I formatted this same USB card reader+SD Card using the FAT32 filesystem and the behaviour was the same as before, only the CH376 could see the files. Also I noticed that the files TEST_0.TXT up to TEST_3.TXT were missing when listing by option 7. On windows nothing was shown. I tried several allocation unit sizes but none worked.

The only part of the example I changed was:

```
      case 49: //1
        printInfo("COMMAND1: Create and write data to files : TEST_N.TXT");    // Create files
        for(int i=0;i<100;i++){
          sprintf(adatBuffer,"%s%d%s","TEST_",i,".TXT");
          flashDrive.setFileName(adatBuffer);  //set the file name
          flashDrive.openFile();                //open the file

          for(int a = 0; a < 1; a++){          //write text from string(adat) to flash drive 20 times
            flashDrive.writeFile(adat, strlen(adat)); //string, string length
          }
          flashDrive.closeFile();               //at the end, close the file
        }
        printInfo("Done!");
        break;
```

I also tried with a bigger SD card (16GB) and also a USB flash drive (8GB) but they didn't give me the option for FAT formatting under Windows.

Wouldn't it be a problem with some setting of the CH376? Maybe for using a FAT32 device it is necessary to ask it to do so with jumper wires or commands?

delgadosouza, 2020-06-01:
I found this user manual:

http://read.pudn.com/downloads766/doc/3040633/CH376_usr_manual.pdf

It is in simple chinese but Google Translate works fine sentence by sentence. It has some code as example, don't know if you have used it already. I am having a look on it to see if I understand something.

trotterlg, 2020-06-01:
A 2 gig card is not a hc card. I would try formatting it with fat16 and see what happens. Larry

delgadosouza, 2020-06-01:
@trotterlg ,Is FAT the same as FAT16?

[image]

There is no FAT16 option, even on command line:
[image]

It reads at the last line "FORMAT is not available for FAT16 units"

delgadosouza, 2020-06-01:
I tried a few things.

1) I used the pendrive with 8GB and in Windows I used the Disk Manager to wipe it and create only one partition with 3GB. The format option was FAT (16/12) and I was forced to use 64K as allocation unit because there is a limitation of 65526 in the number of clusters.
The behavior was:
[image]
Those 2 files I don't know where came from.
And under Windows no file was shown.

2) I used the Disk Manager to wipe it and create only one partition with 2GB. The format option was FAT (16/12) and I could use 32K as allocation unit. I used the 32K option.
After creating 100 files, under option 7 it listed only the first "TEST_0.TXT".
Under Windows it showed only the other 99 files.
Same behabior as with the 2GB SD Card + USB card reader.

3)  Same as 2) but I used the 64K option. The results were like 1), lists 2 strange files and nothing under Windows.

delgadosouza, 2020-06-01:
With the FAT32 option I get always the same behaviour: The files are created, under option 7 only files TEST_4.TXT to TEST_99.TXT are shown and Windows shows nothing.
I tried 2GB and 3GB partition sizes, I tried 8192, 16K and 32K allocation unit size.  Always same behaviour.

trotterlg, 2020-06-01:
Try fat or download SD Formatter and use it.

djuseeq, 2020-06-02:
First of all, thank you guys for testing. Tomorrow I'll borrow one Win10 computer so i can try it. Please can you try to download and update the lib with [this branch](https://github.com/djuseeq/Ch376msc/tree/update) ? Then you will have a new function `getChipVer()` so let's see is it any differ between our modules firmware because that Chinese user manual what you linked before (i have never seen before)  describe the file management procedures differently than in the [English](https://www.mpja.com/download/ch376ds1.pdf) datasheet which i used as a reference. Probably the linked manual is newer than English datasheet, this can also mean that there is a chip on the market with newer firmware that may need to be handled differently, even if it is not mentioned in the data sheet( as many other things have not been described properly) :( , or just the CH chip is choosy for partition type or drive model. That would be the best what @trotterlg mentioned, to use some formatting tool for the flash drive

delgadosouza, 2020-06-02:
Hi, my chip version is 0x42.

I resorted to some Chinese forums using Google translate.
If you use Chrome you can simply right-click on the middle of the page and select "Translate to English". Some people there have the same problem.
I found a library for STM32 in their GitHub clone:
https://gitee.com/think3r/stm32f103_ch376_uart
In this library they state that:

> VERSION NUMBER
> The latest chip version number is: 0x45, successfully tested on win10
> The version number that can only be formatted with xp is: 0x42
> ... ...

It is in the first section of this PDF:
https://gitee.com/think3r/stm32f103_ch376_uart/raw/master/README_Frame.pdf

Other guy used [Diskgenius to check the formatting under Windows 10 and Windows XP](http://www.wch.cn/index.php/bbs/thread-64973-1.html) and the data in the disk is different (have no idea what the numbers are lol):
[image]
[image]

Also apparently [somebody suggested solving this not defining a flag](http://www.wch.cn/index.php/bbs/thread-64784-1.html) "DEF_IC_V43_U":

**yphy0**
> Help!
> 1: The return value of the newly created file in the U disk is USB_INT_SUCCESS, and CH376FileOpenPath( TarName) can be opened, and CH376GetFileSize() can be correctly large
> It's small, but most of the time you just can't see the newly created file on the computer.
> It can be opened and the correct size can be obtained, indicating that the creation was successful. Why can't I see the file on the computer?
> 2: Regardless of whether the file can be seen in the computer, when reading, the return value of CH376ByteRead() and other functions is also USB_INT_SUCCESS, and its RealCount parameter
> The number is also the correct size, but the buf parameter is always an empty string.
> It can be opened and the correct size can be obtained. It should be said that the creation of the file is also successful. The RealCount parameter is also correct. Why does the buf parameter not get the content?
> In addition, the remaining space of the obtained U disk is always correct.

**they**
> Do not define DEF_IC_V43_U in the code. Run the 0x01 command once to check the version number.

**yphy0**
> Finally found CH376DiskQuery( (PUINT32)buf) not running;
> Bewildered. Why doesn't CH376DiskQuery( (PUINT32)buf ); work? And all the returned values ​​are correct.

**Main battle tank**
> I also encountered this problem with the 430 MCU. Later, when I used a chip bought from WCH, this problem did not occur. The chip I had was bought from Taobao.

**think3r** (Probably the guy who created the library mentioned earlier)
> Brother, what is your version number? Mine also encountered the problem of not being able to write to the file.
> My version number is 42H (02H). Formatting under xp or diskgenius will succeed. .

So I guess chip or firmware version is indeed important and I was not lucky.

But I have some hope of it being solved on software because I [found this thread](http://www.wch.cn/index.php/bbs/thread-64697-1.html) where the guy's problem have certainly been solved because he says:

**Jingong Youth**
> The gratitude is beyond words! ! Only 3 points were given, and I couldn’t express my gratitude, so I knelt in front of the screen to reply.
> Reflect on yourself: I don’t know what is eroded by toxic radio waves, confidently believe that my chip is the new version of the firmware, and remove all compatible parts.
> If it wasn't for the master to wake up, he still lives in a dream and debugs hard!
> Bye, bye!

djuseeq, 2020-06-02:
Thanks a lot, this info is very useful. My chip is with the newer version of fw (0x43), so that is the reason why someone have issue with the files opening on PC. I will update my code in a few days but i need your help to test it with your module ;) .

delgadosouza, 2020-06-03:
Hi, you can count on me!

delgadosouza, 2020-06-03:
> Try fat or download SD Formatter and use it.

Hi @trotterlg it indeed worked with the FAT format, thanks for the idea. Apparently it works only with 32K (or maybe less) allocation unit in that format. I didn't try but from the facts I believe it will work as well for FAT32 if using Windows XP or a third-party formatting software.
The only problem for me is that having to use third-party software is bad because I intend to distribute my project to people that are not very keen on computers. The best scenario would be them to buy a new flash drive and use it with no hassle.
For now my conclusion is that I will have to buy the chips with the most updated firmware I can find.

djuseeq, 2020-06-07:
Hi @delgadosouza , if you can please download and try the [updated](https://github.com/djuseeq/Ch376msc/tree/fix_oldFW) lib. Be aware , this is not tested because i don't have suitable chip, so I have no chance to try it

delgadosouza, 2020-06-30:
Hi @djuseeq , I am sorry for the long wait, I had a deadline in another project so didn't have time to test the library.

I downloaded the updated lib (Ch376msc-fix_oldFW.zip file) and tried it with two flash drives:

 - 8GB formatted with FAT32. The files created under Windows 10 didn't show up in command 7 (list). The file created with command 1 didn't show up in Windows 10.
 - 2GB formatted with FAT. Both ways have worked fine.

Just reporting other subject, I have some odd behaviour when listing files after creating folders. I noticed that:

**With the 8GB FAT32 flash drive**
when I do the sequence create file > create folder > list files I get
```
************************
COMMAND7: List directory
************************

/.          
/..
```

Also the command 3 (read file) don't show anything.
If I create the file again and try to list again the I get:
```
************************
COMMAND7: List directory
************************

/.          
/..         
TEST1   TXT : 1940 >>>	1.9 Kb

```
and I can read the file.

**With the 2GB FAT flash drive**
In a freshly formatted drive when I list the files I get:
```
************************
COMMAND7: List directory
************************

SYSTEM~1    : 0 >>>	0 byte

```
Creating and listing the file works fine BUT, when I open/create folders (command 9) and do the listing (7) I get:
```
************************
COMMAND7: List directory
************************

```
The file disappears. The file and folders show under Windows. Even if I remove and insert the drive again nothing changes.
Now if I create the file again and try listing I get:
```
************************
COMMAND7: List directory
************************

SYSTEM~1    : 0 >>>	0 byte
TEST1   TXT : 1940 >>>	1.9 Kb
/DIR1
```

Strange behaviour... I checked both versions of the library, the fix_oldFW branch and the master branch, same behaviour. Maybe it is a problem with the example code?

In the special commands did you find now info regarding the problem with FAT32?

Thank you for the help!

djuseeq, 2020-07-10:
>I am sorry for the long wait, I had a deadline in another project so didn't have time to test the library.

Absolutely no problem. In this case I'm afraid I can't help because I don't have a similar chip on which i could test the code.
I would come back to it in the future when i get a similar chip with old firmware.
The chip handles FAT32 internally, we just instruct what to do with the file. In some functions, the instructions are different between the old and new version chips, for this I found information in the given code.

kaernter53, 2021-02-15:
Hi djuseeq,
I spent 2 days testing the interface with different USB sticks. Here is my result - some are not recognized at all, regardless of the capacity. Others are recognized by the interface, you can write and read, but windows shows an empty directory. I found 2 pieces that work as desired. 1. Sandisk Cruzer Blade 16GB and 2. Transcent 4GB. Both support USB 2.0. On the Cruzer Blade I reduced the partition to 3.64GB and formatted it with FAT32. I left the Transcent unchanged, it is also formatted with FAT32. I have now ordered 2 Sandisk 16GB and will then try to see if they work with 16GB. I will report here. My firmware is 066.

djuseeq, 2021-02-16:
Hi @kaernter53 .
I think your module has the old firmware (if 066 is decimal number = 0x42 hexa). They are not working properly yet. If you have a chance to format the USB stick with some Linux, Win XP or Win 7, that would be great.

kaernter53, 2021-02-17:
Hi djuseeq,
thanks for the answer. Du you know what's the aktual release number an where i get a "new release" module??

djuseeq, 2021-02-17:
According to manufacturer the new releases are 0x45 (69) and 0x44 (68). Mine is with version 0x43 (67) . I'm afraid you have to find it, because unfortunately the sellers don't even know which version they have. I ordered from [here](https://www.aliexpress.com/item/32664654472.html?spm=a2g0s.9042311.0.0.27424c4d8N0VpR) couple of years ago, but that doesn’t have to mean you will get the same as me.

djuseeq, 2021-03-18:
Master branch and Readme is updated. Please read the [Known Issues](https://github.com/djuseeq/Ch376msc#known-issues) and the [test](https://github.com/djuseeq/Ch376msc#firmware-difference) sections. ;)

closed 2021-03-18 by djuseeq