djuseeq/Ch376msc #30: File Creation date wrong

gcharles81, 2020-07-19:
Hello Thanks for this library I am using it in my Aquarium controller project but I have an issue on teh creation date of the files created on the USB somehow the date shows always 31.12.2003 time 23.00 is there a reason for that ?

Thanks 
Charles 
[image]

djuseeq, 2020-07-20:
Hi Charles.
I'm glad you can use it, thanks for the star. The default date/time it should be 2004-jan-1, 0h.0m.0s . What is the process during creating the RTCLOG.TXT file? First create and save the file,after set the file name again and open it, using the setYear, setMonth ... functions set the actual date/time, after call saveFileAttrb function and finally close the file. Actually with these functions you change the file "last modified time and date" attributes. Maybe i will add a possibility to choose between create and modified date attributes to change.

gcharles81, 2020-07-20:
Hello Thanks for the Answer , so the solution to that is to use the below functions at startup and always update those with Today`s date ? I have RTC so I can keep thoose updated or I have to update before editing any file ?

          flashDrive.setYear(2019);
          flashDrive.setMonth(12);
          flashDrive.setDay(19);
          flashDrive.setHour(03);
          flashDrive.setMinute(38);
          flashDrive.setSecond(42);

djuseeq, 2020-07-20:
If the file doesn't exist on the drive then first you create it, if there is some data then write it, close the file with this you will create the file on the drive 
`setFileName(FILENAME); openFile(); writeFile(data, length); closeFile();` this will create the file with the default date 2004-jan-1, 0h.0m.0s
and then open again the file and modify the date
`setFileName(FILENAME); openFile(); get the date/time from RTC and use the set functions  ; saveFileAttrb(); ` after these you can close the file or continue to add a next line of data and close it afterwards `moveCursor(CURSOREND);  writeFile(data, length); closeFile();`

closed 2020-08-28 by djuseeq