djuseeq/Ch376msc #31: CH376 with SPI

AbdelazezMRX, 2020-08-06:
i am using Arduino Mega With CH376 using SPI, but when i see the data frame using logic analiser it seems that i have no response,
just the same data i sent using SPI is return back again, i don't know if the problem in the connection or what , but i tried many configurations to the pins and still not working , and i need it because it is part from my graduation project.
i hope to help me fixing the problem, and thanks
[image]

djuseeq, 2020-08-06:
Hi, so there is no response at all from the module? While sending data to the module, is it normal if you get back the previously sent byte from modules shiftregister , the arduino will discard it, so when arduino wants to read data from the module , it would send 0x00 to the module while in the same time receives the useful data from the module.

AbdelazezMRX, 2020-08-06:
thanks for your reply,
but when i send 0x00 to module to get the useful data i just get the previous byte i had send, i mean the module doesn't respond with any data, it is just work as a shift register, i don't know if the problem in the connection or what, because the datasheet says to make TX/RX HIGH when using SPI and make A0 also HIGH, and this is different from the picture in your library.

djuseeq, 2020-08-06:
Ok, if you use this library then you don`t need to send any data directly to the module, the lib will do that. You have to check the connection between the module and the Arduino board and jumper config. If your module differs from my one, then its better to inspect with DMM or just check the printed text on the module and pull to ground RD and WR pins and pull it to HIGH(probably 3.3V) the PCS pin. Check the TXD pin, if its pulled to the ground with jumper then remove that jumper (serial mode).
[image]

closed 2020-08-28 by djuseeq