djuseeq/Ch376msc #32: writeFile(buff, len) doesn't work reliably.

ohbhatt, 2020-09-10:
Files are left with a single character written. Logically implementing the same code using writeChar(char) gives the desired result.
Felt I should through some light on it.

djuseeq, 2020-09-16:
Hi, that is weird, please can you share more info about your project, i mean using SPI or UART(hw or sw?), speed? Do you use some interrupt in your code? Have you tried the example code? If yes the it have the same behavior? Thanks

ohbhatt, 2020-09-16:
Thank you for your response. I am using Hardware Serial @ 115200 Baud. I have tried the example code and it works well most of the time. Sometimes, it only writes one character to the file. I do use interrupts in the rest of my code, A GPS uses another Hardware Serial port, here this problem is more frequent.

I tried writing the same file again and the frequency of the problem has reduced quite a bit, but it still happens 1/5 times.
Please let me know if another Serial connection could hinder with the functions of the library.

My particular code is very large, but I have exactly replicated what is done in the example.

djuseeq, 2020-09-29:
Sorry but i can't reproduce it. Can you try over SPI? Check the wires and try just with Arduino board without attaching any other modules on it with the example ino file.

ohbhatt, 2020-10-01:
Sure I can do that. I'll get back to you soon with my result.

closed 2024-10-21 by ohbhatt