djuseeq/Ch376msc #37: I keep getting "Attach Flash Drive First" message

cosmicone, 2020-11-06:
After connecting to Arduino as described in the "SoftwareSerial" example sketch,

I get the menu -- but when I type any input, it says "Attach Flash Drive First" 

I have tried three different USB Flash Drives and same error appears.

djuseeq, 2020-11-07:
Hi, probably physical connection and/or speed config problem. SoftSerial is mostly for testing, it have his own [limitations](https://github.com/djuseeq/Ch376msc/issues/15)

cosmicone, 2020-11-08:
What is the default baud rate on hardware serial 1 of Arduino Mega 2560? How can I match the two baud rates between CH376S and Arduino Mega 2560?

Meins321, 2020-11-11:
Well with the "softserial" example i get the red led to light but not the usb stick to be recognized on my Ch375B Chipped board...

maybe we need the CH37**6** because those have apparently build in FAT16/FAT support? i don't know why but usb stick write/read on arduino plattform is a mistery now for me i pretty much own all of the usb host boards none of them works with any stick

[image]

Meins321, 2020-11-13:
Okay problem found!
CH376S Module works directly :1st_place_medal:  aye

09:07:07.941 -> 
09:07:07.941 -> 
09:07:07.941 -> ****************************************
09:07:07.941 -> h:Print this help
09:07:07.941 -> 
09:07:07.941 -> 1:Create
09:07:07.941 -> 2:Append
09:07:07.941 -> 3:Read
09:07:07.941 -> 4:Read date/time
09:07:07.941 -> 5:Modify date/time
09:07:07.941 -> 6:Delete
09:07:07.941 -> 7:List dir
09:07:07.941 -> 8:Print free space
09:07:07.941 -> 9:Open/Create folder(s)/subfolder(s)
09:07:07.974 -> ****************************************
09:07:07.974 -> 
09:07:08.143 -> Flash drive attached!

Meins321, 2020-11-13:
> 
> 
> Hi, probably physical connection and/or speed config problem. SoftSerial is mostly for testing, it have his own [limitations](https://github.com/djuseeq/Ch376msc/issues/15)

first of all it could be the CH37**5** module ? ^^ :-7

cosmicone, 2020-11-13:
> 
> 
> > Hi, probably physical connection and/or speed config problem. SoftSerial is mostly for testing, it have his own [limitations](https://github.com/djuseeq/Ch376msc/issues/15)
> 
> first of all it could be the CH37**5** module ? ^^ :-7
> 
> Second: Details of the USB Stick?
> IS it FAT32 formatted and how big? 1/2/4/8GB with FAT32 works directly no problem!
> 
> I own both CH Modules.. can you give me your mail adress?
> i can send you the "other" version if you need help to support CH375B better?

I tried three different modules and 3 different USB sticks - each in SW serial, HW serial and SPI. I get the same message on all.

The USB sticks are FAT32 formatted on Linux (avoided Windows 10)

cosmicone, 2020-11-13:
> 
> 
> Okay problem found!
> CH376S Module works directly 🥇 aye
> 
> 09:07:07.941 ->
> 09:07:07.941 ->
> 09:07:07.941 -> ****************************************
> 09:07:07.941 -> h:Print this help
> 09:07:07.941 ->
> 09:07:07.941 -> 1:Create
> 09:07:07.941 -> 2:Append
> 09:07:07.941 -> 3:Read
> 09:07:07.941 -> 4:Read date/time
> 09:07:07.941 -> 5:Modify date/time
> 09:07:07.941 -> 6:Delete
> 09:07:07.941 -> 7:List dir
> 09:07:07.941 -> 8:Print free space
> 09:07:07.941 -> 9:Open/Create folder(s)/subfolder(s)
> 09:07:07.974 -> ****************************************
> 09:07:07.974 ->
> 09:07:08.143 -> Flash drive attached!

Okay - what was the problem you found? More details please. What can I do to make this work?

djuseeq, 2020-11-14:
> What is the default baud rate on hardware serial 1 of Arduino Mega 2560? How can I match the two baud rates between CH376S and Arduino Mega 2560?

Mega doesn't have a default baud rate, You have to initialize the MCU UART interface with appropriate value (e.g. 9600, 19200, 57600, 115200 ), this is done by the library when you do initialization (Ch376msc(Serial, 9600) where Serial is a valid port(e.g. Serial, Serial1, Serial2, ...) ), please try to use the module on default [9600](https://github.com/djuseeq/Ch376msc#getting-started)  bps, and connect (Mega_TX - CH_RX, Mega_RX - CH_TX) or try over SPI interface.

djuseeq, 2020-11-14:
> first of all it could be the CH37**5** module ? ^^ :-7
>
> Second: Details of the USB Stick?
> IS it FAT32 formatted and how big? 1/2/4/8GB with FAT32 works directly no problem!
>
> I own both CH Modules.. [...]

Yes is it possible that 375 chip differs from 376, sadly i don't own 375 module so I have no idea what is the difference. 
That would be really helpful to improve the library. You can find it [here](https://github.com/djuseeq/Ch376msc/blob/master/library.properties) . Some of the CH376 chips still have old Firmware and requires different sequence of commands to getting work, I'm waiting for one of this module, hopefully it will arrive soon then i can start to do some improvements. [OldFW](https://github.com/djuseeq/Ch376msc/issues/28)

ArcRobotics, 2022-06-18:
Hi i am having the same issue and i have ordered 2 nodules they are working at all i even at some point connected them with a ttl converter and nth happened the module contains CH375 chip what should i do?

ArcRobotics, 2022-08-03:
> first of all it could be the CH37**5** module ? ^^ :-7
>
> Second: Details of the USB Stick? IS it FAT32 formatted and how big? 1/2/4/8GB with FAT32 works directly no problem!
>
> I own both CH Modules.. [...]

can u send me the CH375 version my email adress is omaralrafei.95@gmail.com

agganitk, 2023-10-11:
please let know the solution.  I am also keep seeing same error  "Attach Flash Drive First". I am using Arduino Uno.

LewiLim, 2024-04-05:
> Okay problem found! CH376S Module works directly 🥇 aye
>
> 09:07:07.941 -> 09:07:07.941 -> 09:07:07.941 -> **************************************** 09:07:07.941 -> h:Print this help 09:07:07.941 -> [...]

what was the solution please?