djuseeq/Ch376msc #38: H/W Serial on Zero/M0 ZERO ATSAMD21G18 = Attach flash drive first!

Meins321, 2020-11-11:
Please add what D1 does deafult wise?

Does it light up when initializind the module is done? while reading it does what?

Okay so i did wire it up like suggested in the readme for my Arduino Zero and Hardware Serial

(MCU) --(Module)
(PIN0) -- (TXT)                 //alway cross TXD RXD
(PIN1) -- (RXD)
(D5)     -- (GND) 

Should give 115200 baudrate detached later for default baudrate. JUMPER left to Serial not Parallel

#####
Test (1)
Ch376msc flashDrive(Serial1, 115200); //Ground Wire from D5 to GND on MCU Board
SerialUSB.begin(9600);
Outcome:
i do get the Serial output and can access the menue nothing more..
USB Stick is not recognized (4GB FAT32)
########
Test (2)
Ch376msc flashDrive(Serial1, 9600); // NO WIRE ON D5 Module - GND MCU
SerialUSB.begin(115200);

power off, usb stick attached
Red LED D1 constant lit after powercycle?
Still cant read or do anything :-/

Meins321, 2020-11-11:
/*------------------------------------------------------------------------------------------------------------------
 *    Author: György Kovács                                                                                         |
 *    Created: 28 Mar 2019                                                                                          |
 *    Description: Basic usage of CH376 with hardware serial                                                        |
 *    Thanks for the idea to Scott C , https://arduinobasics.blogspot.com/2015/05/ch376s-usb-readwrite-module.html  |
 *------------------------------------------------------------------------------------------------------------------
 */

#include <Ch376msc.h>

//..............................................................................................................................
// Leave the default jumper settings for the baud rate (9600) on the CH376, the library will set it up the chosen speed(HW serial only)
Ch376msc flashDrive(Serial1, 9600); // Ch376 object with hardware Serial1 on arduino mega baudrate: 9600, 19200, 57600, 115200
//..............................................................................................................................
 // buffer for reading
char adatBuffer[255];// max length 255 = 254 char + 1 NULL character
//..............................................................................................................................
// strings for writing to file
char adat[]="Vivamus nec nisl molestie, blandit diam vel, varius mi. Fusce luctus cursus sapien in vulputate.\n";
char adat2[] = "Lorem ipsum dolor sit amet, consectetur adipiscing elit. Duis efficitur ac est eu pharetra. \n";
//..............................................................................................................................
unsigned long totSect = 0;
unsigned long freeSect = 0;
byte percentg = 0;
byte tmpCommand; //used to store data coming from serial port
boolean readMore;
static char helpString[]= {"h:Print this help\n\n1:Create\n2:Append\n3:Read\n4:Read date/time\n"
            "5:Modify date/time\n6:Delete\n7:List dir\n8:Print free space"
            "\n9:Open/Create folder(s)/subfolder(s)"};

void setup() {
  SerialUSB.begin(115200);
  flashDrive.init();
  printInfo(helpString);
}

void loop() {
	if(flashDrive.checkIntMessage()){
		if(flashDrive.getDeviceStatus()){
			SerialUSB.println(F("Flash drive attached!"));
		} else {
			SerialUSB.println(F("Flash drive detached!"));
		}
	}
  if(SerialUSB.available()){
    tmpCommand = SerialUSB.read();                      //read incoming bytes from the serial monitor
    if(((tmpCommand > 48)&&(tmpCommand < 58)) && !flashDrive.driveReady()){ // if the data is ASCII 1 - 9 and no flash drive are attached
       printInfo("Attach flash drive first!");
      tmpCommand = 10; // change the command byte
    }
     switch (tmpCommand) {

      case 49: //1
        printInfo("COMMAND1: Create and write data to file : TEST1.TXT");    // Create a file called TEST1.TXT
          flashDrive.setFileName("TEST1.TXT");  //set the file name
          flashDrive.openFile();                //open the file

          for(int a = 0; a < 20; a++){          //write text from string(adat) to flash drive 20 times
            flashDrive.writeFile(adat, strlen(adat)); //string, string length
          }
          flashDrive.closeFile();               //at the end, close the file
        printInfo("Done!");
        break;
//*****************************************************************************************************************************************************
      case 50: //2
        printInfo("COMMAND2: Append data to file: TEST1.TXT");               // Append data to the end of the file.
        flashDrive.setFileName("TEST1.TXT");  //set the file name
        if(flashDrive.openFile() == ANSW_USB_INT_SUCCESS){               //open the file
        	flashDrive.moveCursor(CURSOREND);     //if the file exist, move the "virtual" cursor at end of the file, with CURSORBEGIN we actually rewrite our old file
        	//flashDrive.moveCursor(flashDrive.getFileSize()); // is almost the same as CURSOREND, because we put our cursor at end of the file
[... 158 lines cut ...]
        break;
    }//end switch

  }//endif serial available

}//end loop

//Print information
void printInfo(const char info[]){
  
  int infoLength = strlen(info);
  if(infoLength > 40){
    infoLength = 40;
  }
    SerialUSB.print(F("\n\n"));
    for(int a = 0; a < infoLength; a++){
      SerialUSB.print('*');
    }
   SerialUSB.println();
   SerialUSB.println(info);
   for(int a = 0; a < infoLength; a++){
      SerialUSB.print('*');
    }
   SerialUSB.print(F("\n\n"));
}

Meins321, 2020-11-11:
My programm code in the arduino ide is just the basicUsageHwSerial.ino

i changed only every "Serial." for "SerialUSB." as i am on a Zero that is needed to get the Programming Port (MicroUSB) to post text to the "Serial Monitor"

Line 14: Ch376msc flashDrive(Serial**1**, 9600); // at all times just changed the baudrate and single ground wire on the data lines

Meins321, 2020-11-11:
okay Line14 Ch376msc flashDrive(Serial1;*Baudrate*); seems to fail for anything that is not 9600 baud then the red led is lit constantly with or without usb stick :-/

while no jumper wire is on the Datalines from the module..

Does the baudrate setting not work on the SAMD21 maybe?

Meins321, 2020-11-11:
Btw maybe add a link to the arduino serial overview from here to the compatability lists for beginners?
https://www.arduino.cc/reference/en/language/functions/communication/serial/
If you test this library with an uno and hardware serial :-) it collides as there is only one availible

djuseeq, 2020-11-14:
Hi, the D1 LED on CH37**6** module is lit when the initialization is done (USB HOST mode) and if USB Device is attached.  After power cycle the LED must to be in off state if initialization is not issued . CH37**6** chip on Arduino Zero HW serial is not tested, but works over SPI, in your case CH37**5** doesn't have SPI interface :( . If You have then please try on Mega board, CH375 is compatible with CH376 but still can have a minor differences in a command structures (i don't have it so i cannot test it :( . )

djuseeq, 2020-11-14:
> [image]

Is this your module in this photo or was it taken from the internet? Since the crystal does not appear to be of adequate value, 12 MHz is required, not 2 MHz

Meins321, 2021-03-19:
The picture is my module i usally not take other pics :-P

the model are obtained from german amazon...

if you tell me your adress i will ship it with padded envelope to you to get support rolling as a donation for this awesome library!

moritz dot m AT gmx DOT de

djuseeq, 2021-03-21:
Thanks for the offer and for Your support, I am happy to accept it. Mail has been sent ;)

djuseeq, 2021-04-01:
Thanks for your generosity, now i was able to test the library with CH375. Sadly but i have to correct myself, the CH375 don't use the standard ( 1 start bit, 8 data bit , 1 stop bit ) serial configuration, it use 9 data bit instead. I was confused by the messy datasheet :  :(
> "The USB-DEVICE Mode of CH376 is totally compatible with CH372, the USB-HOST Mode of
CH375 is basically compatible with CH375."

 Probably they meant CH376 is basically compatible with CH375. Not really, maybe the commands but not the serial communication protocol , maybe the parallel. Grrrrr 

The good news is that the ATmega328 and ATmega2560 support's 9 data bit mode, and the bad news is the Arduino HardwareSerial library doesn't. So I need to find a solution to stay with the simplicity of the Arduino and allow this way of communicating with the MCU.