djuseeq/Ch376msc #41: Deleting a file in a sub-directory deletes said directory

burnt-past, 2021-01-22:
Using the snippet below deletes the entire  _DIR2_ subdirectory.

Looking that the datasheet it seems that the file needs to be _opened_ before deleting, otherwise the entire directory is deleted. I'm not sure if this would be classified as a bug, since deleting a folder has the same command (as far as the CH376s is concerned). Maybe adding some info to README.md would be helpful for others.

I spent a bit tearing my hair out trying to figure out what was going on. Hope this helps others.

**Deletes Sub-Directory**
```
flashDrive.cd("/DIR1/DIR2", 0);
flashDrive.setFileName("TEXT.TXT");
flashDrive.deleteFile();
```

**Deletes File**
```
flashDrive.cd("/DIR1/DIR2", 0);
flashDrive.setFileName("TEXT.TXT");
flashDrive.openFile();
flashDrive.deleteFile();
```

djuseeq, 2021-01-22:
Thanks for your time and effort, i will look on it.

djuseeq, 2021-03-18:
Code is updated, there is a new function as well, called deleteDir() . Try it and if is still persist a problem, feel free to reopen this issue. ;)

closed 2021-03-18 by djuseeq