djuseeq/Ch376msc #43: Can it open .bin file and read byte ?

gcharles81, 2021-02-28:
Hello thanks for the library and I use it in few of my projects to write text files and have no issues 

Today I wanted to experiment if possible to use a USB drive to update Arduino directly from USB but I am not sure if Library can open .bin files and Read one byte at a time , Is that possible with the library.

The sd Library has has a function called readBytes

Regards
Charles

djuseeq, 2021-02-28:
Hi, i have to make a PR and update the master branch, but please try [this](https://github.com/djuseeq/Ch376msc/tree/rawRead). [#35 Here](https://github.com/djuseeq/Ch376msc/issues/35) we talked about this. ;)

djuseeq, 2021-03-18:
The master branch updated, I hope you can achieve what you want ;)

closed 2021-03-18 by djuseeq

gcharles81, 2021-04-08:
Hello @djuseeq I used the master branch and I can open the .bin files , I could proceed implementing  the function of updating an arduino Due firmware via a usb stick - thanks for this great library