djuseeq/Ch376msc #45: File not created on my pendrive but read works on serial monitor

Aravinthkalai1, 2021-03-22:
It's working fine on serial monitor ever functions works fine but when I eject Pendrive and insert it into my pc there are no files created! am using windows 10 64bit please help me to sort out the issue.

djuseeq, 2021-03-22:
Hi. Yes, that is a [known problem](https://github.com/djuseeq/Ch376msc#known-issues) with modules which have a chip with obsolete firmware. Please check your chip's firmware version, calling getChipVer() function and print it on serial terminal. Please read the possible solution [here](https://github.com/djuseeq/Ch376msc#firmware-difference) .

Aravinthkalai1, 2021-03-23:
It's working fine. I just formatted my Pendrive with Active@ Partition Manager, thank you for your guidance.

closed 2021-03-23 by Aravinthkalai1