djuseeq/Ch376msc #47: Writing on file slowly!

etoshey, 2021-03-31:
Hi,
Thanks for your useful library.
I am trying to write my sensors data on file, but there is a long delay after each writing block (256 bytes), So I have to wait for the next ready(Done) from Ch376 for more than 3 milliseconds.
How much is the maximum speed for writing a file?

best regards.

etoshey, 2021-04-03:
Thanks for your reply. I have done all of them. I am using SPI communication with 12MHZ SCK and every 128 bytes is written at 4 milliseconds. Is it normal?

djuseeq, 2021-04-07:
Yes, that is normal ,we have to wait with the next command until the CH376 finish with file writing otherwise data loss will occur. The chip will inform the MCU when is ready to receive the next command or data. The most time consuming part is when writing continuously and the CH376 internal buffer(512byte) is filled up, then the chip require longer time to do some internal processes, this is much depends on the usb-drive used.  [Here](https://github.com/djuseeq/Ch376msc#firmware-difference) you can find more details about the drives i have been used.
2GB no name drive, worst case is more than 91 ms waiting for the module.
[image]

32GB PNY, worst case is 3.5 ms waiting for the module.
[image]