djuseeq/Ch376msc #53: [SPI interface] USB is not detecting and Prints as Attach flash drive first!

embhobbb, 2021-07-04:
I am using Arduino Uno borad and CH376S USB host module. And I have connected with SPI interface as shown in the following figure.
I didn't connect INT pin.
[image]

I was trying basicUsageSPI example but didn't work for me. On serial it was printing like as 
```
*************************
Attach flash drive first!
*************************
```
Any suggestions?

djuseeq, 2021-09-15:
Sorry for the late reply, is there something else connected to the SPI port or just the module? If the SPI port is shared with other devices too then INT pin is required.

ymg2006, 2021-12-11:
Same problem here. The SPI port is not shared at my setup. The only connected device is CH376s. Tested with Arduino UNO and WEMOS Lolin32. The UART is working fine but SPI is not working on both. The light on the CH376s stays off when the USB is inserted but the LED on the memory itself blinks when inserted. 
Here is a complete picture of the working and not working setup:

Color        CH376s      Arduino UNO     Name
Yellow          D3                 D10               SCS                  
Blue             D5                 D11               SCK
Purple          D6                 D11               SDI-MOSI 
Pink             D7                 D11               SDO-MISO
White          5v                  5v                  5v
Black           GND              GND              GND 

Used your default example [basicUsageSPI](https://github.com/djuseeq/Ch376msc/blob/master/examples/basicUsageSPI/basicUsageSPI.ino).

[image]

[image]

[image]

I think the chip is fine because it functions properly on UART mode.
Could the CH376s chip be partially not working? how to test this?

ymg2006, 2021-12-11:
After checking wires, one of them was corrupted, After getting it fixed and it worked well.
I think you should close this issue. I will verify on WEMOS Lolin32 too. Thanks.