djuseeq/Ch376msc #55: Arduino Due not connecting to CH376

sidprice, 2021-09-13:
Hello, I have the CH376S module and I have installed this library. My board is an Arduino Due and I have followed the wiring diagram on this repo. However,my Arduino Due is not able to communicate with the module.

I have done some debugging I I find the library stays in the loop shown below in the "CheckIntMessage" function:

		} else {//spi
			while(!digitalRead(_intPin)){
				tmpReturn = getInterrupt(); // get int message
				delay(10);//sadly but it required for stability, sometime prior attaching drive the CH376 produce more interrupts
			}// end while
		}//end if interface

The library is initialized with:
      ` Ch376msc flashDrive(10); // chipSelect`

Advice on getting this working would be much appreciated.

djuseeq, 2021-09-15:
Hi. I know about it, i have [tested](https://github.com/djuseeq/Ch376msc#tested-boards) and in SPI mode works just with using an extra pin (interrupt). Use `Ch376msc flashDrive(10, 9); // chipSelect, interrupt pin` instead.

sidprice, 2021-09-15:
Thank you I was able to get it to work using the interrupt input. However, I need to have multiple SPI devices on the bus so I do not think I can use this method.

Do you know if the CH376 works if not using the INT input?

Many thanks for the help :)

sidprice, 2021-09-15:
Seems I spoke too soon. While the menu appears on the serial monitor (basicUsageSPI sketch) the CH376 does not access the flash drive. It just asks for a drive to be connected.

sidprice, 2021-09-21:
I am still unable to access my flash drive. I have hooked up the INT input as you sugegsted and also I have formatted the drive with Active@ Partition Manager.

I would appreciate some help in getting this to work.

djuseeq, 2021-09-22:
> Thank you I was able to get it to work using the interrupt input. However, I need to have multiple SPI devices on the bus so I do not think I can use this method.
> 
> Do you know if the CH376 works if not using the INT input?
> 
> Many thanks for the help :)

Yes, this is the only option if you plan to attach multiple devices to the SPI bus.  MISO, MOSI, SCK signals are shared between devices, every device has his own CS line and CH376 requires plus an extra pin for INT signal.

djuseeq, 2021-09-22:
Check that the communication between the DUE and the module is correct, use `getChipVer();` method to query the program version. What value did you get?

trotterlg, 2021-09-22:
I have used a solution to this issue that works well for me. I use two logic level shifting boards that have an enable pin on them. Tie one side together and use the remaking two sides to go to your devices. Then just enable the one you want with it enable pin while disabling the other.

trotterlg, 2021-09-22:
[image]

sidprice, 2021-09-23:
Many thanks for the suggestions and help. The issue was the hardware configuration, the module documentation and the chip datasheet are difficult to understand. After digging out my USB to SPI monitor (SPIDriver) and just try lots of simple Python tests I was able to discover the correct hookup.

closed 2021-09-23 by sidprice