djuseeq/Ch376msc #64: file name Limitations

clear-sky, 2022-01-14:
hi dear György Kovács
and thanks for your nice code!
a question!
why file name must have 8 capitalized char ?
is there any way to solve this Limitations?
as far as i know sd cards dos not have this limitation.

djuseeq, 2022-01-18:
Hi, @clear-sky . The chip are support the so-called MS-DOS-compatible short file name in 8.3 format. It is also possible to implement long filename support, but I do not find information on this in the incomplete (poorly written) documentation

djuseeq, 2022-03-18:
That will not works in this way :( . I will check the source you attached.

alvarotorijano, 2022-12-08:
Hi is there any update with this? This chip seems to be the first choice to be used with all the arduino family

petervanderwalt, 2022-12-09:
Hi! 

Any chance to have Long Filename Support?   https://openbuilds.com/threads/interface-cnc-touch-controller.16906/page-11#post-129152 refers

```
The CH376 does support long file names. It is specifically mentioned in the specs. It may just be a limitation for writing files.

"Provide file management functions: open, create or delete files, enumerate and search files, create subdirectories, support long file names;"

Their doc specifies an example "EXAM1" for long file names, which I cannot seem to find on their site...

```

goguelnikov, 2023-05-27:
this could be a very interesting addition to the library, indeed. 
Any chance there is some good news about it?
Best