djuseeq/Ch376msc #65: ch376s not working with esp32

microElabDevelop, 2022-01-29:
i try to run ch376 use module with SPI interface , i m using esp32 controller, i try but not succeed, i littel bit consue about connection hai jumper setting related to spi.  plz give me connection diagram and, example code.

WR > GND
RD > GND
CS > 3V3
D7 > pin_spi_miso= esp32 ->D19
D6 > pin_spi_mosi= esp32 ->D23
D5 > pin_spi_sck   = esp32 ->D18
D3 > pin_ch376_cs= esp32 ->D5
INT > pin_ch376_int =NOT USE
Ch376msc flashDrive(5);
I USE THIS SETTING.

djuseeq, 2022-01-29:
Hi. It was discussed here #18 . Do you have attached other device as well to the SPI? Try using with INT pin involved. `Ch376msc flashDrive(5, INTPIN);`

microElabDevelop, 2022-01-29:
No i m only try to run ch376 with spi.
I read the link. 
Can u explain  me 
Fs=usb
And 
Spi speed.?

djuseeq, 2022-01-29:
The default SPI speed is 125kHz. If you want to use higher speed then you can choose from two macro for that when you instantiate CH object. In a kilohertz range use e.g `Ch376msc flashDrive(5, INTPIN, SPI_SCK_KHZ(500) );` ,this will have the SPI clock at 500kHz, or in megahertz range use `Ch376msc flashDrive(5, INTPIN, SPI_SCK_MHZ(8) );`  for 8MHz.

microElabDevelop, 2022-01-31:
thanx for helping me , problem is that , im confusing in spi mode or parallel mode,, Tx jumper remove from GND .

closed 2022-01-31 by microElabDevelop