djuseeq/Ch376msc #69: Accessing USB Pendrive and SD Card consecutively!

shivang705, 2022-05-07:
Hello @djuseeq

Thanks for providing this insane library! 
I followed up the steps provided for interfacing SD card on the module and VOILA! just setSource(1) and it works!
Now, I want to access both USB Pendrive and SD Card consecutively. 

Here is what I did - 
1) Used the basicUsageHwSerial example code
2) Added 8 more cases - A,B,C,D,E,F,G and H which replicas of existing first 8 cases - 1,2,3,4,5,6,7,8
3) In each added case, in the start of the case, setSource(1)
4) In the end of the case, setSource(0)

Outcome -
1) Shows "Flash Drive attached!" and "Flash drive Detached!" on insertion/removal of pendrive.
2) Able to write, append, read and delete files on pendrive.

Issues I am facing for SD card - 
1) I am writing in the file, but while reading it, shows empty.
2) While appending to the file. it shows "Disk Full" message.

So is it possible to use both consecutively using setSource() function or is it required to reinit the IC for SD mode and USB Host mode?

Also, when I was working only on SD Card, I found out that the files on the sd card are not visible on File Explorer in Windows 10! Is there any specific reason for this or am I doing something wrong?

Thanks in advance!

djuseeq, 2022-05-10:
Hi @shivang705 , thanks. 
The `setSource()` function will initialize the IC to SD or USB Host mode. If i understand correctly, you want to use the SD card and Flash Drive simultaneously, what is not supported by the IC. This behaviour with the SD card is similar to IC with old FW and Win10 formatted partition. Try to format with third party software.

shivang705, 2022-05-10:
Thanks for the response!

I referred the datasheet and found out that power-on / external reset is mandatory before switching to SD mode. I will try doing that!

I will also try formatting the SD card with third party software and will test it again! 

Also, when both USB pendrive and SD card are connected, with setSource(0) at the beginning of the code, USB pendrive works fine. But, with setSource(1) at beginning of the code, SD card does not works. It always shows Attach drive first.
Is it because of init() function where IC is set to default  MODE_HOST_0 mode?

djuseeq, 2022-05-10:
You don't need to reset the module, just call `setSource()` function with appropriate mode (0 - USB, 1 - SD). First always init the module, this will configure to MODE_HOST_0 and afterwards you can switch to SD. Also very important to use the capacitors + resistor in the SD card VCC line because otherwise when you inserting the card that will cause a large voltage drop for a short period on the VCC line which leads to instability issue and this can cause what you noticed(get the "Attach drive first" message)  and if this happens then the only solution is resetting the module.

shivang705, 2022-05-23:
Sorry for the late reply!

Right now, I have connected both SD Card and USB pendrive. I have set setSource(1) with SD mode. But I am unable to read/write through SD Card. 
But when I remove Pen Drive, It works. And when I reconnect, it doesn't. Why is it so?

djuseeq, 2022-06-12:
It can be a voltage instability on the VCC line. 

> Also very important to use the capacitors + resistor in the SD card VCC line because otherwise when you inserting the card that will cause a large voltage drop for a short period on the VCC line [...]

shivang705, 2022-06-16:
Ok! Thanks for the response! Now, I have connected external power supply (SMPS) to power the IC and it is working great!

closed 2022-06-16 by shivang705