djuseeq/Ch376msc #70: Init function hangs the application

msufian9, 2022-06-19:
It would be great if the init() function return something.....
because on startup/setup code, init() is called as per the suggestion,  init() should be called before anything....
There is a scenario where the ch376 module fails, then the whole system will fail..
So it would be great if init() function return 0 of 1 showing weather the device is initialized successfully or not .

djuseeq, 2022-06-21:
Hi. did you use on UART or SPI interface? Hangs mean doesn't return from init function? This can be a good approach but in this moment you can verify the module with calling `pingDevice()` or `getError()` function.

msufian9, 2022-06-22:
I have used SPI interface with ESP32....
Hags mean the programme stop at the point where we are calling Init(). So the whole firmware stops there and the device does not work at all