djuseeq/Ch376msc #73: i can't write string to flash drive.

SaimShuja, 2022-08-30:
i have used most code from [basicUsageSoftSerial](https://github.com/djuseeq/Ch376msc/blob/master/examples/basicUsageSoftSerial/basicUsageSoftSerial.ino)  example .

my actual issue is i want to write a String to file periodically.
but the Ch376msc::writeFile(char* buffer, uint8_t b_size) expects a char array .
so i have tried converting my string to char array but the file dosent get written, it is created but it dosent open.
i have tried three different ways but still couldnt write String.

method 1:
`int n = inString.length();
     char char_array[n];
   strcpy(char_array, inString.c_str());`
    

method 2:
`int n = inString.length();
     char char_array[n];
    inString.toCharArray(char_array, n);
`

method 3:
`flashDrive.writeFile((char *) inString.c_str(), inString.length()); `

but only file is created i can see it in my computer but i am unable to access it.

is there a proper way to write string to attached usb.

whole function for refrence:

````
void store(String inString)
{ // write data read on Serial to the Usb storage

  String name = date + ".txt";       // genreate filename using current date
  flashDrive.setFileName(name.c_str()); // set the file name
  uint8_t res = flashDrive.openFile();

  Serial.println("[response] " + String(res));

  if (!res == ANSW_ERR_MISS_FILE) // open the file
  {
    flashDrive.moveCursor(CURSOREND); // if the file exist, move the "virtual" cursor at end of the file, with CURSORBEGIN we actually rewrite our old file
    //flashDrive.moveCursor(flashDrive.getFileSize()); // is almost the same as CURSOREND, because we put our cursor at end of the file
  }

  if (flashDrive.getFreeSectors())
  { // check the free space on the drive

    /*
    might be a problem here
    need to convert string to char array
    lets see if this helps
    */

    int n = inString.length();
    // char char_array[n];

    // /* following are two different ways to conver string to char array */
    // // string to char array using stcpy
    // // strcpy(char_array, inString.c_str());

    // // string to char array using tochararray method
    // inString.toCharArray(char_array, n);

    // // for debug purpose
    // Serial.println("[String] " + inString);
    // Serial.print("[char *] ");
    // Serial.println(char_array);
    // flashDrive.writeFile(char_array, n); // string, string length
    flashDrive.writeFile((char *) inString.c_str(), inString.length()); //string, string length
  }
  else
  {
    Serial.println("Disk full");
  }
  flashDrive.closeFile(); // at the end, close the file
  Serial.println("Done");
}
````

msufian9, 2022-09-03:
You have to convert the string to char array.
Try making extension ".txt" to .TXT"

msufian9, 2022-09-03:
*****Try this code snippet*****
flashDrive.setFileName(nowFile.c_str());
flashDrive.openFile();
String saveToFlash="";      
int strLen = saveToFlash.length() + 1;
char buf[strLen];
saveToFlash.toCharArray(buf, strLen);
flashDrive.writeFile(buf, strlen(buf));
flashDrive.closeFile();