djuseeq/Ch376msc #88: Unable to get the SPI interface to work

Java6502, 2024-07-29:
I have seen a few other posts regarding SPI issues; however, the resolution doesn't help my issue.  I have the device working in serial and parallel modes, with this API, as well as ones I wrote myself.  The problem is that I just can't get any communication to work over SPI.  I have the device connected exactly as depicted in your diagram.  
Thanks,
Jonathan

Jeronimo59, 2025-05-06:
Hello @Java6502,

After some troubleshooting, I managed to get it to work using SPI.

First step for me was to tweak the SPI port setup until the pingDevice() function worked.

FYI, since my SPI was also used by another device, I had to extend the library with an additional constructor:
```
   Ch376msc(SPIClass *spi, uint8_t spiSelect, uint8_t intPin, SPISettings speed = SPI_SCK_KHZ(125));
```

Let me know if I can help.

Java6502, 2025-05-10:
Thanks for the reply.  I pulled the device out and hooked everything back up.  Just to be clear, the connections you have shown on the right (on your Github README) are for SPI.   

MISO, MOSI, SCK, and SS are connected, as well as all of the pins in red????  I have no other devices connected to the UNO, and so I am simply using the following.  Here is a minimal test just to see if pingDevice works.

#include <Ch376msc.h>

void setup() {
  Serial.begin(115200);
  Ch376msc usb(10);  
  usb.init();
  usb.pingDevice();//THIS ALWAYS FAILS
}

void loop() {
}

The call to usb.pingDevice() always works with UART (different code from what's above of course), but for some reason fails with SPI.  I added a Serial.println in the source code of pintDevice just to see what spiReadData() returns.  For some reason I always get 0x55.

Please clarify the connections for SPI.  Are all of those ground and 3.3v connections necessary?  Also, I am not sure what is meant by tweaking the SPI port setting.  Should I use a different clock speed?  Anyway, thanks for any help you can offer.

Jeronimo59, 2025-05-10:
> Just to be clear, the connections you have shown on the right (on your Github README) are for SPI.

Indeed, the connections in blue are for SPI (pls nbote that since I am not @djuseeq , it is not "my" Github README ;-)

> Please clarify the connections for SPI. Are all of those ground and 3.3v connections necessary? 

Indeed all 5 connections in red are necessary as they indicate to the chip it needs to operate in SPI and not serial.
As suggested, I also used another GPIO to connect to INT (cf green indication), so I used the constructor Ch376msc(uint8_t spiSelect, uint8_t intPin, SPISettings speed). You may want to try that configuration.
As I am using an ESP32, I am using other GPIO pins that the UNO's so I cannot reproduce your exact setup.

> The call to usb.pingDevice() always works with UART (different code from what's above of course), but for some reason fails with SPI. I added a Serial.println in the source code of pingDevice just to see what spiReadData() returns. For some reason I always get 0x55.

I did the same, and I played around with cabling (I didn't need to change the SPI speed) until spiReadData() returned 0xFE as expected.

HTH